open-rdc/orne_navigation #80: 横断歩道対策

Nasupl, 2015-07-13:
* 追従どうする？
など

Hiroki-Goto, 2015-08-23:
今回出たこと

追従組
　・ROSの読み込み
　・スイッチ1つでモードチェンジ（？）←横断歩道とそれ以外のモード（？）←　座標値でモードチェンジ
　・横断歩道をどうわたるか（段差がきつい）

白線処理
自己位置でやるか？
waybpint　navigationでやるときついかも
地図にはセルごとに属性値と尤度は与えられるけどどうするか・・・
セルに情報入れると管理しずらい
地図とエリア情報を一括で管理する
これらは orne　navigationにパッケージしていく
FSM のパッケージにあるかも
目標値をロボットの少し後ろにしていれば障害物回避を回しながら追従行動できるかも

走行関係
モータのトルク関係？　
タイヤを大きくする？　←段差考えるとこっちかも
鬼畜どんぐり　←大きく浮かんだりする

Nasupl, 2015-08-24:
追加・補足メモ（？）
* 横断歩道前や対象発見時のPathPlannningについて
	* 状態管理するノードを追加するべき
	* 一つのノード内でやる
	* 地図を分割する？
		* 地図内でエリアを分割する
		* orne_navigation内にパッケージを作るべき
	* ロボットの手前に目標を定めれば障害物回避しない
		* nav_goalをロボットの手前に設置してあげる

ShunShimamura, 2015-09-15:
@Hiroki-Goto 状態管理関係
@Shiina-Shiotani  地図関係
@ShunShimamura 追従関係

buzzshima, 2015-09-19:
座標値よりも地図を分割して状態を管理していったほうがいいかも
今回は地図と状態管理がmast

buzzshima, 2015-09-19:
追従関係
前のロボットを認
waybpointナビゲーションするのは追従までの地点まで
そこの区域に入ったら障害物があれば少し前のところをゴールにしておく

Nasupl, 2015-10-19:
#87 は完了
横断歩道関連では #104 が進行中

Nasupl, 2016-06-14:
https://github.com/DaikiMaekawa/fulanghua_navigation の開発によって，横断歩道手前での一時停止は可能になった．今年度は信号機のある横断歩道に取り組むため一旦閉じる．

closed 2016-06-14 by Nasupl