open-rdc/orne_navigation #82: マッピング失敗の原因調査

Nasupl, 2015-08-24:
7/4の試走会でマッピングに失敗した原因を調査する
* icart_miniｎROS Packageを利用している他の大学で同様の事例が発生したとの報告がある
* 複合オドメトリに問題が発生している
 * 当日のbagファイルの検証を行う

ShotaroTerato, 2015-09-08:
複合オドメトリに問題は見受けられなかった

DaikiMaekawa, 2015-09-08:
@ShotaroTerato そこが一番可能性がありそうだったけど、どうやって調べたの?

DaikiMaekawa, 2015-09-08:
Issueには検証方法とその結果を書いておくと後で見返した時に便利だと思いますよ。

ShotaroTerato, 2015-09-13:
問題のロボットの進む方向が反転する時のbase_link: pose: pose: position: x,yの値の挙動を監視したところ，ロボットの進む方向が反転しても，x,yの値の増減が反転するということは起こりませんでした．

DaikiMaekawa, 2015-09-13:
@ShotaroTerato  /combined_odomとかいうトピック名だったと思いますがそこのthetaの値も見ておいた方が良いです。

Nasupl, 2015-09-13:
私も同じデータを確認しましたが，/combined_odomトピックを見る限りでは姿勢が180度転換するようなデータは流れていないように見えました．gmappingがpublishしてる~entropyトピックを見てみて，これが高くなっていたらやはりgmappingが怪しいのかもしれません．

DaikiMaekawa, 2015-09-13:
そうですか。 /combined_odomが問題ないのであればgmappingを調べる必要がありそうですね。 
私は直接現象を見ていないので何とも言えませんが以下のケースが近いでしょうか?

http://answers.ros.org/question/12446/turtlebot-slam-map-building-problems/

だとすればこの記述は重要そうですね。

> My sensor/power board has the ADXRS620 gyro and not the ADXRS613, the big difference is that ADXRS613 has a +/- 150deg/s output and the ADXRS620 has a +/- 300 deg/s output. This is important because for the same angular speed the raw outputs will differ by a factor of two.

DaikiMaekawa, 2015-09-13:
あっ、でもよく読んだらこのQ&AはTurtlebot固有っぽいな。

DaikiMaekawa, 2015-09-13:
トラブルシューティングの項目は確認済みですか?

http://wiki.ros.org/navigation/Troubleshooting

DaikiMaekawa, 2015-09-13:
姿勢が突然反転したとのことですが、問題発生時のタイムスタンプ付近のtfで変換エラーが発生してないかも見ておく必要があると思います。

Nasupl, 2015-09-16:
@DaikiMaekawa 
tfの変換エラーは発生していません．トラブルシューティングに該当項目は確認できませんでした．

gmappingのentropyトピックはfixedな値しか出ていないので，参考になりませんでした・・・．
rqt_consoleでメッセージを見てみると，問題が発生する直前にimuがAngle change too lateのエラーを吐いているので，これが間接的に原因になっているのかもしれません．ただし，１８０度回転する前後数秒で発生しているわけではないので，直接的な原因は不明です．

DaikiMaekawa, 2015-09-19:
私にbagファイルを渡して下さい。 解析してみます。

ShotaroTerato, 2015-09-20:
@DaikiMaekawa 
bagfileを添付したメールをyahooのアドレス宛に送信しました．
ご確認お願いします．

DaikiMaekawa, 2015-09-25:
長らく放置しており申し訳ありません。 今時間が取れそうなのでちょっと解析してみます。

DaikiMaekawa, 2015-09-25:
だいたいわかりました。 動画を見やすく編集してから状況をまとめますのでもう少し待っていて下さい。 後、bagファイルがあれば容易に状況を再現することが出来ますが実際の動きと比較することはその場にいた人しか出来ないと思うので、動作時の様子を記録することを忘れないように。

DaikiMaekawa, 2015-09-25:
https://www.youtube.com/watch?v=yujwZr9nsaM&feature=youtu.be

/combined_odomは不連続に生成されており、それが地図生成に悪影響を与えているのが見て取れますね。 ロボットを回転させた時に少しズレた地図が生成されるのはこれが原因のようです。 また、姿勢が大きくズレる前に姿勢が大きく回転しているのがわかります。 左のrqt_consoleの値は少し遅延しているのですが、IMUの値が波打つように変動しており、それがオドメトリにも反映されています。

DaikiMaekawa, 2015-10-17:
修正完了。

closed 2015-10-17 by DaikiMaekawa