open-rdc/orne_navigation #83: 機体ごとのlaunchやparamの管理

Nasupl, 2015-08-24:
マシンごとのlaunchファイルや各種パラメータの差異はlaunchファイルのif-unlessルールを使って管理し切り替えられるようにする
http://wiki.ros.org/ja/roslaunch/XML

DaikiMaekawa, 2015-09-11:
加えて、icart_miniのリポジトリじゃなくて、orne_navigationのリポジトリに配置すること。 waypoints_cfgもこっちに持ってきていいと思いますよ。

ShotaroTerato, 2015-09-21:
icart_miniのリポジトリの方にadd_hokuyo3dブランチを作成しました．

DaikiMaekawa, 2015-09-21:
終わったらプルリクエストを投げること。 私がレビューします。

ximo112, 2015-09-29:
@DaikiMaekawa 
icart_miniリポジトリのselect_robotブランチでロボットの指定をできるようにしました、プルリクエストを送りましたのでレビューよろしくおねがいします。

Nasupl, 2015-10-02:
* icart_mini_drive.launchにデバイス名が直に書いてある
 * if-unlessルールを利用して，指定したマシンのデバイス名が読み込まれるようにしてください
* icart_mini.paramにもマシンごとの差異が存在する
 * αとβそれぞれのパラメータを作成して，icart-mini.shに引数を与えてマシンを指定するとそれぞれのparamが読み込まれるようにする必要がある

DaikiMaekawa, 2015-10-02:
icart_mini.paramもdebリリースするためにパッケージ化しました。 https://github.com/open-rdc/orne_navigation/tree/indigo-devel/orne_setup 

マシン毎のパラメータの違いはここを修正してください。

ximo112, 2015-10-06:
add_sensor_selectでロボットごとのデバイス名、
add selection paramでparamの指定をできるようにしました。
add_sensor_selectではα側のデバイス名が分からなかったので中身の同じlaunchファイルが存在しています、with_hokuyo3dファイルは @Nasupl 中身の変更をおねがいします。

DaikiMaekawa, 2015-10-16:
https://github.com/open-rdc/orne_navigation/pull/94 にて対応した。 10/17日の試走会で動作確認をしてもらう。

DaikiMaekawa, 2015-10-17:
masterにマージした。

closed 2015-10-17 by DaikiMaekawa