open-rdc/orne_navigation #84: CI

DaikiMaekawa, 2015-08-24:


DaikiMaekawa, 2015-08-24:
継続的なビルドテスト・単体テストの自動化は非常に有意義である。
GitHubならtravisが最もお手軽だけど、Ubuntu 14.04に対応していないので indigo, jadeのテストが困難。
そこで、Jenkinsを研究室サーバで稼働させることを検討する。

DaikiMaekawa, 2015-08-24:
サーバ環境が既にあるのでJenkinsの導入自体は難しくありません。 

@Nasupl 一日あれば出来ると思うのですがどう思います?

DaikiMaekawa, 2015-09-08:
@Nasupl Jenkinsサーバの動作を確認しました。 他の人にも引き継げるようにここのWikiに立ち上げ方を書いといて。

DaikiMaekawa, 2015-10-16:
Jenkinsにログインしていろいろと作業していたのですが、かなり動作が不安定です。 突然ログイン出来なくなったり、プラグインのインストール中でエラーを出して停止するなど。

サーバーが手元に無いのであまり詳しい情報も得られず、使い勝手が悪いですね...

DaikiMaekawa, 2015-10-16:
そんな中、travis-ciがついに14.04のβサポートを始めました。 Jenkinsは便利なツールですがこちらに移行する方が良いかもしれません。

http://blog.travis-ci.com/2015-10-14-opening-up-ubuntu-trusty-beta/

DaikiMaekawa, 2015-10-18:
[icart_mini](https://github.com/open-rdc/icart_mini)の方でTravisをサポートしました。 orne_navigationはTravis導入前にimuパッケージのimuノード... (ADISのROSドライバ)の扱いを何とかしなければ

DaikiMaekawa, 2015-10-20:
https://github.com/open-rdc/adis_imu_drivers ADISのリポジトリにもTravisを導入しました。 プルリクエスト毎に自動でテストが実行されることを確認。 テストが失敗した場合はGitHub側がそれを認識し、プルリクエストはmasterにマージできなくなります。

Nasupl, 2017-05-30:
閉じます

closed 2017-05-30 by Nasupl