open-rdc/orne_navigation #90: Testing management script on the real robot

DaikiMaekawa, 2015-09-29:
他プロジェクトでこちらのソースを流用することになったので、細かい箇所を微調整しています。 https://github.com/open-rdc/icart_mini/pull/78 の修正ではicart_mini_driver.launchを立ち上げた際に従来使用していたypspur_coordinatorの起動スクリプトも同時に立ち上がるようにしました。 今までの状態だとパッケージとしてスクリプトを管理出来ませんので他の用途に流用する際に色々と面倒でした。

ブランチを変更してから実機でテストをお願いします。

DaikiMaekawa, 2015-09-29:
@Nasupl 取り敢えず担当者を夏迫君にしましたが、誰でも良いので実機で実行してみてください。 今日は開発日だと思いますので、出来れば今日中に結果を書き込んでもらえると助かります。

Nasupl, 2015-09-29:
@DaikiMaekawa
今日はもう帰宅してしまったので明日午後イチで実験します．申し訳ないです．

ximo112, 2015-09-29:
ORNE_betaにてrun-coordinator-processブランチのicart_mini_driver.launchを試しました。

icart_mini_driver.launch
[image]

./icart-mini.sh
[image]

---------------
ctrl+cをした時
icart_mini_driver.launch
[image]

./icart-mini.sh
[image]

DaikiMaekawa, 2015-09-29:
@ximo112  ./icart-mini.shは必要ありません。 この修正でicart_mini_driver.launchを実行した際にセットで起動するようになりました。 一番目のスクリーンショットを見る限り起動には成功しているようですので試しに何かしらの指令を送ってみて実機が無事動いたらOKです。

ximo112, 2015-09-29:
teleop_joy.launchにて正常な動作を確認。

DaikiMaekawa, 2015-09-29:
協力に感謝します。

closed 2015-09-29 by DaikiMaekawa