open-rdc/orne_navigation #104: Erroneous detection problem at the intersection

DaikiMaekawa, 2015-10-18:
交差点に存在する勾配で障害物を誤検出し自律走行を継続できない問題が発生している。 バグファイルを解析し、対策を考える。

DaikiMaekawa, 2015-10-19:
以前３次元の姿勢データをmove_base側に反映させる修正をした記憶がありましたので簡単なテストをしてみました。 姿勢が傾いている状態では地図との整合性を保つように障害物の登録が抑制されます。 

https://github.com/open-rdc/adis_imu_drivers/pull/19 の修正をマージすれば実機でも同様の効果が期待出来ます。 以前は通信の問題があったので使用するデータをヨー軸だけに制限していました

[image]

DaikiMaekawa, 2015-10-19:
@open-rdc/project-orne 実は校内でも勾配がキツくて同様の現象を再現出来る箇所があります。 新２号館を出ていつもロボットをスタートさせる地点から６号館(?)までの経路だったと思います。 他にも探せばいろいろあると思いますので急勾配を登らせたり、それを含むコースの地図を作成して全体を通して自律移動のチェックをしたりと本件に関しては念入りに実機検証してください。

DaikiMaekawa, 2015-10-19:
何か問題があった場合は直ぐに報告することを心がけてください。 そろそろ時期的に考えても、対応が少し遅れただけで致命傷になりかねませんので。

DaikiMaekawa, 2015-10-20:
@Nasupl こちらの件、報告お願いします。

Nasupl, 2015-10-22:
@DaikiMaekawa 
短距離なら問題ないものの、長距離走行するとIMUの蓄積誤差のためか姿勢がめちゃくちゃになってコストも正常に付与されなくなります
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/

DaikiMaekawa, 2015-10-22:
@Nasupl どうやらそれは既知のバグのようです。 IMUの周期を計算するところのバグにより、蓄積誤差が貯まるようです。 
https://github.com/citbrains/electronics/pull/13

IMU_robocupが修正版のようですので試して下さい。

DaikiMaekawa, 2015-10-22:
@Nasupl あと、もう一つ大きな問題としてはこちらですね。 https://github.com/open-rdc/cit_adis_imu/issues/26

幸い修正は簡単そうです。

Nasupl, 2015-10-22:
@DaikiMaekawa 
IMU_robocupの最新のIMUのコードについてですが，周期の変更に伴って正常に角度が出力されません．IMUの内部での積分周りのパラメタの問題かとは思うので調整してみていますが，いまいちうまく行きません．
@yasuohayashibara 
Robocupの方ではIMUから出力される角度情報は使用していないようですが，この件について何かご存知でしたらご教示ください．

yasuohayashibara, 2015-10-22:
ロボカップでは角速度データをベースに制御していたので，IMUから出力されている角度データに関してはあまり検討されていません．ロボカップ内ではこれに関してはほぼ未検証です．

DaikiMaekawa, 2015-10-22:
@Nasupl @yasuohayashibara 相変わらずこのインタフェースボードには手を焼かされますね。 内部の角度データが信用出来ないのであればロボカップと同様に角速度データをPC側で積分して使用しますか?

以前切り替えた際に使い物にならなかった主な理由としては通信エラーとIMU側の周期の問題だと思いますので今ならもう少しマシな結果が得られるのではないでしょうか。

DaikiMaekawa, 2015-10-23:
去年と同じ時期に、同じくIMUの問題に悩まされるというのは面白いですね。 https://github.com/open-rdc/orne_navigation/issues/41

過去ログから前回の議論を参照しましたが、データ量が増えると取りこぼし頻度が上がるようですので角速度情報を吐き出すようにインタフェース基板側のコードを修正するのであれば角度は切ったほうが良さそうです。  念の為報告しておきます。

> @Nasupl 
IMUから送られてくるデータに角速度情報を追加してからは、データの取りこぼしの頻度が上がっています。
IMUとの通信頻度を下げるとデータが安定することからデータ量の増加が原因と考えられる。
やりとりするデータを3軸の角度データのみに限定し、チェックサムを追加して下さい。

DaikiMaekawa, 2015-10-23:
https://github.com/citbrains/electronics/pull/9 こちらの修正はrobocup_IMUにあるソースコードには含まれていないと推測します。 必要な場合は適宜つくばチャレンジ側で使用していたソースコードをマージして使用してください。

DaikiMaekawa, 2015-11-03:
@Nasupl こちらの完了したと考えてよろしいですか?

Nasupl, 2015-11-03:
@DaikiMaekawa 
歩道から横断歩道へ下る際の問題は解決しましたが、横断歩道から歩道へ登る際は、横断歩道の下り勾配がそれほど大きくないため、歩道が障害物としてコストに登録され、歩道に登ることはできませんでした。

DaikiMaekawa, 2015-11-03:
@Nasupl 写真または動画をよろしくお願いします。 対策方法を早急に考え出します。 
最悪の場合、URGの取り付け高さ変更、地図・ウェイポイント共に作り直しの可能性もあります。 リプライを急いでください。 :cry:

Nasupl, 2015-11-03:
@DaikiMaekawa 
状況としては先日Youtubeにアップした動画と全く同じ状態でした。今回の動画は残念ながらありません。
bagファイルは取ったのでアップします

DaikiMaekawa, 2015-11-03:
最悪のケースに備えて、URGをどの高さまで上げれば障害物として登録されずに走破できるのかデータを取っておくべきかもしれませんね。 これは時間との兼ね合いになってくると思いますが。

Nasupl, 2015-11-03:
設計データと路面データ( http://www.tsukubachallenge.jp/news2015/2015/04/24/3907 )からある程度は検討ができるかもしれません

DaikiMaekawa, 2015-11-03:
その路面データは有意義ですね。 私は設計データが手元にありませんので見積に関してはそちらにお任せしてもよろしいですか?

Nasupl, 2015-11-04:
マッチングとは別に，コスト付加用のURGを新たに取り付けるのはどうでしょうか

DaikiMaekawa, 2015-11-04:
URGの余裕があるのであればそれでも構いませんが。

DaikiMaekawa, 2015-11-04:
差し当たり、local_plannerのmin_obstacle_heightパラメータを変更してみるのはどうでしょうか。 現在、地面と同じ高さ0.0に設定されていますが障害物に合わせて変更してみるのもありです。

DaikiMaekawa, 2015-11-04:
> URGの余裕があるのであればそれでも構いませんが。

後はUSBポートの余りがあるかどうかも関係しますね。 パラメータを調整してみてダメだったらURG追加も検討しましょう。

パラメータに関しては以下を参照してください。
http://wiki.ros.org/costmap_2d/hydro/obstacles

Nasupl, 2015-11-04:
βに関して言えば，研究室に1台Ether-URGが余っていますし，取付け部品も全て揃っています．αに関して言えば，3D URGから出てくるポイントクラウドをある平面だけ取り出してコストマップに渡してあげれば解決できます．
min_obstacle_heightの調整で何とかなる気もしますが，横断歩道から歩道への乗り上げについては学内で検証できないため，現地での調整になります．
現地での調整でうまく行かない場合に備えて，URGの増加も並行して実施します．

DaikiMaekawa, 2015-11-04:
@Nasupl よろしくお願いします。 因みに路面データもあり現地でどの程度の角度かも確認しているとは思いますが学内で再現出来ないのはどういった理由からでしょうか? 模擬した環境物を簡易的でもいいので作成し検証することは可能ですか? 幸いそれぐらいの人手は確保出来そうに見えますが。

DaikiMaekawa, 2015-11-04:
学内では開発ではURGの追加を実施する前にパラメータの調整を行って下さい。 結果が乏しくない、やり方がわからない場合は質問していただければ出来るだけ早めに回答します。

Nasupl, 2015-11-04:
@DaikiMaekawa 
模擬的にあの段差を再現するには，資材の購入，組み立て，用意に1日かかると思います．明後日にはまた現地での実験が始まるので，厳しいのではないでしょうか．

DaikiMaekawa, 2015-11-04:
ロボットの姿勢変化に対して同じ効能が得られれば良いのです。 以下私の案です。

1. ロボットに直進するだけの経路を生成させ、追従させる。 (nav_static_mapでよい)
2. 段差昇降時に機体が傾く高さ分だけ手で機体を傾ける。 (事前に作ったガイドを使用する)
3. Costmap2Dの挙動を監視し、パラメータを修正する。
4. 1に戻る

DaikiMaekawa, 2015-11-04:
当然プランニングに衝撃等は使用していないため検証したいのは、あくまでも直進したい時に何度までピッチ方向の傾きに対応できるかどうかということに問題を分解出来ると考えました。 

機体を何度まで傾けるのかという指標を人間にわかりやすくするため簡易的なガイドだけ作成し、傾けるのは人手で行えば良いのではないでしょうか。 ガイドは棒でも何でも良いです。 もちろんこの方法では論文などで使用できる程の再現性はどうやっても得られないと思いますが、今関心がある事柄はそこではないと思います。

Nasupl, 2015-11-04:
ガイド等は設けず，パラメタを変更して簡易的に実験しました．
min_obstacle_heightを0.2にすれば，先日の21°の勾配（高さ18cm）の前で10°程度傾けてもコストは付加されませんでした．これは，角度等を厳密に検証していないので，つくばと完全に同条件という訳ではありませんが，ほとんど同条件となっていると思います．

DaikiMaekawa, 2015-11-04:
逆に元のパラメータではコストが負荷されるということも同条件で確認をお願いします。 また、安全のためパラメータは実際よりは少し大きめに設定しておいてください。 検証方法とパラメータの最適化に関してはお任せします。

ximo112, 2015-11-05:
新しくwaypointを追加するという手を使えば、新しいwaypointまでのpathは通り、また、新しいwaypointの地点からであれば前方のcostは無くなり現在のwaypointまでのpathも通ると思います。
[image]

ximo112, 2015-11-05:
この方法であると、新しく追加するwaypointがcostに近いため、どの程度costに近い状態でwaypointの最短pathが通るか確認したほうが良いと思います。costの沼の中にwaypointを置いてみるなど...

Nasupl, 2015-11-05:
パラメータの最適化についてですが、URGの取り付け高さが地上から0.2mに設定されているため、min_obstacle_heightを0.2mより高く設定すると、平面を走行している際にも障害物に対してコストが付加されない現象が起こりました。
今回の実験では、対象とする段差の高さが0.2mだったため、0.2m前後が最適でしたが、現地の段差は、最大の高低差は0.2m近くあるものの、コストに影響を与える高低差は0.15m程度であるため、これを最適なmin_obstacle_heightとして設定することとしました。

Nasupl, 2015-11-05:
また、 https://github.com/open-rdc/orne_navigation/issues/104#issuecomment-153962416 のWaypointの追加について追記すると、現在のWaypointは道路側と歩道を登りきった側のみに設置されており、その間で段差を検出してコストの壁ができるために、ローカルなパスを生成できずに停止してしまっている状態であると考えられます。それに対して、コストの壁の手前かつ段差を登った直後にWaypointを置けば、横断歩道から新たなWaypointまでの間は問題なく経路の生成が行われ、かつそのWaypointを通過する段階ではコストの壁が消えた状態になるため、問題なく横断歩道を通過できるというものです。

DaikiMaekawa, 2015-11-05:
:+1: いいと思います。 min_obstacle_heightは更なる保険として有効ですのでこちらもプルリクエストしていただけると助かります。

DaikiMaekawa, 2015-11-05:
こちら動作確認が完了しましたら閉じて下さい。

DaikiMaekawa, 2015-11-06:
#104 を追加したことにより交差点の段差で障害物を誤認識する問題が解消されることを試走会にて確認。

closed 2015-11-06 by DaikiMaekawa