open-rdc/orne_navigation #113: AMCLのパラメータ調整

Nasupl, 2015-10-23:
* チューニング前にランドマークが開けた場所でロボットを走行させてパーティクルの広がりを監視したほうがいい
* 障害物や自己位置のズレてパーティクルの広がりがあまりに早い場合はオドメトリの信頼度を上げる

iou16, 2015-10-23:
@DaikiMaekawa オドメトリ信頼度を上げるということですが、これはサンプリングステップの誤差パラメータ(具体的にはamclのodom_alpha1, odom_alpha2, odom_alpha3, odom_alpha4)を低く設定することでパーティクルの広がりを抑えるようにするという形でよろしいでしょうか？

DaikiMaekawa, 2015-10-25:
はい、差し当たりはその方法でノイズの期待値を下げて試してみてください。 こういったパラメータの調整は結構泥臭い作業なので動作中やバグファイルで挙動を何度も確認し、必要に応じて他の該当パラメータもいじる必要が出てくるかもしれません。 とにかく検証実験は数をこなすしかありません。

iou16, 2015-10-27:
パラメータ調整の現状を報告いたします．
*今回は，公園内においてスタート地点(テント地帯から見て右・トイレ付近)の直線からスタートし，そのまま開けた場所(公園の道路側・地図に記録されるような障害物なし)に突入した場合に発生したパーティクルの発散を学内の環境で擬似的に再現し検証しました．
* odom_alpha4(回転方向の誤差を基準に回転方向の分散を決めるパラメータ)を低くする(信頼度を上げる)と，パーティクルの分散が抑えられる印象があります．ですが，誤差が蓄積すると復帰に若干の時間がかかる印象があります．
* odom_alpha3(並進方向の誤差を基準に並進方向の分散を決めるパラメータ)を低くする(信頼度を上げる)と，あまりいい結果が得られません．もともとオドメトリの誤差が大きいため真の姿勢に対するパーティクルのの喪失が発生するため誤差は多めに見積もって置くことが必要だと思われます．
* パーティクルが大きく分散する原因として，前後方向の誤差が発生した状態(amclの姿勢の推定値と真の姿勢の誤差が発生している状態)で旋回することが挙げられます．再現するのが若干難しいのですが，つくばで発生したのと非常に酷似したパーティクルのばらつき方をします．
* @DaikiMaekawa 質問なのですが，odom_alpha1とodom_alpha2は直接的でない(例：並進方向の偏差が回転方向の分散に影響する)パラメータですが，このパラメータはどのような基準で決定なされたのでしょうか？今後調整するにあたって参考にさせて頂きたく思います．

DaikiMaekawa, 2015-10-27:
@iou16 実際に様子を見ていないので正確なアドバイスは出来ませんが、

> odom_alpha4(回転方向の誤差を基準に回転方向の分散を決めるパラメータ)を低くする(信頼度を上げる)と，パーティクルの分散が抑えられる印象があります．ですが，誤差が蓄積すると復帰に若干の時間がかかる印象があります．

誤差が蓄積するということはそのパラメータはそれ以上下げるべきではないと思います。 実は誤差を大きめに見積もる分には通常大きな問題を引き起こすことはありませんが、誤差を過小評価することは推定結果に大きな悪影響を与えます。 なので、大きめの誤差から徐々に下げていき値を調整するのがコツです。 それに付け加えると、パラメータのチューニングの際には入力データをbagファイルに記録しフィルタに繰り返し流し込みながらRVizでパーティクルの変化を見極めるのが効率的です。 (既に行っていると思いますが念の為)

> パーティクルが大きく分散する原因として，前後方向の誤差が発生した状態(amclの姿勢の推定値と真の姿勢の誤差が発生している状態)で旋回することが挙げられます．再現するのが若干難しいのですが，つくばで発生したのと非常に酷似したパーティクルのばらつき方をします．

動画とbagファイルが欲しいですね。 オドメトリの計算部分を今年変更しましたので、どの要素がシステムに影響を及ぼしているのか見極めたいです。

> @DaikiMaekawa 質問なのですが，odom_alpha1とodom_alpha2は直接的でない(例：並進方向の偏差が回転方向の分散に影響する)パラメータですが，このパラメータはどのような基準で決定なされたのでしょうか？今後調整するにあたって参考にさせて頂きたく思います．

odom_alpha1に並進方向の偏差は関係ないように見えますが。
odom_alpha2はデッドレコニングで発生したドリフトのパラメータのことです。

現状報告から推測するに観測に関するパラメータであるlaser_*に関しても修正が必要になってくるかもしれません。

iou16, 2015-10-29:
@DaikiMaekawa アドバイスありがとうございます、参考にさせていただきました．
* 調整した結果は，odom_alpha1=0.2，odom_alpha2=0.2, odom_alpha3=0.8, odom_alpha4=0.1となりました．調整した印象としては誤差をこれ以上過小評価するとAMCLによって推定される姿勢がが真の姿勢に復帰するのに時間がかかるような印象があります．なので，これ以上サンプリングの誤差パラメータを低くするのは，あまりいい結果が得られないのではないかと考えます．パーティクルのばらつき具合ですが，既存のパラメータよりはパーティクルが円弧状に大きく広がりにくくなっています．
*  前後方向の誤差が発生してる状況で旋回するとパーティクルの収束が遅延する件についてですが，何度か試したのですが，前回報告したような現象を再現することができませんでした．私の勘違いだったのかもしれません申し訳ありません．
* laser_関連のパラメータの調整ですが，確かにサンプリングの誤差パラメータを下げて，パーティクルの喪失問題及び，パーティクルの収束の遅延を発生させる危険を犯すより，誤ったランドマークとのマッチングによってパーティクルがそちらに引っ張られることを防ぐ方が正しいのかもしれません．その場合は，どのようにパラメータを調整すべきかご意見を頂けると幸いです．(そもそも，z_hit, z_short, z_max, z_randを調整するという形になるのでしょうか？)

DaikiMaekawa, 2015-10-30:
こちらでいろいろと回答しました。　[image]

DaikiMaekawa, 2015-11-01:
https://github.com/open-rdc/orne_navigation/pull/118 にてmasterにマージされたためこちらのIssueは閉じます。

DaikiMaekawa, 2015-11-08:
2016年度に向けてパラメータの最適化を再開。

Nasupl, 2016-06-14:
@iou16 本件どういう結論になったんでしたっけ

iou16, 2016-06-15:
小手先のパラメータ調整ではどうにもならないので，別途2D LIDARに匹敵する高精度なセンサを積んで自己位置推定に反映するのがベストな対処法だと思われます．

Nasupl, 2016-06-21:
GPSの統合等は研究で進んでいるので一回こちらは閉じます

closed 2016-06-21 by Nasupl