open-rdc/orne_navigation #114: リカバリ動作の検証

Nasupl, 2015-10-23:


DaikiMaekawa, 2015-10-25:
何をどのように検証するのか書いておくことをおすすめします。 @ximo112

ximo112, 2015-10-29:
2号館3階でnav_static_map.launchを実行し、経路を遮るようにしてロボットを囲みrecoveryを動作させました

*bagfile
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/orne%20recovery%20verification%20.bag

*動画
https://youtu.be/TDekGy7BvFs

DaikiMaekawa, 2015-10-29:
@ximo112 動画を確認しました。 waypoints_navは一定時間毎にゴールポジションをmove_baseに送るようにしていますので、そちらで試していただいてもよろしいですか? 去年はこの修正を入れることでリカバリ失敗後も動き続けるようにしてしました。

ximo112, 2015-10-29:
waypoints_navの動作になります

*bagfile
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/orne_recovery_verification_(waypoints_nav).bag
*動画
https://youtu.be/4a6RyapT3hs

DaikiMaekawa, 2015-10-29:
問題無さそうですね。 @ximo112 何か気付いた点などありますか?

ximo112, 2015-10-29:
waypoints_nav時の動作は特に問題は感じませんでした。
少し気になる点として、突然前方に障害物が現れた際に少しバックすることがあったのですが、これはローカルコストマップ上でコストの沼に入った時、後方にコストが無いと判断された場合に起こる動作なのでしょうか？

DaikiMaekawa, 2015-10-30:
後方移動の優先度は低いですが、0には設定していないので前方の候補が全て無くなった場合にバックする挙動は発生します。

DaikiMaekawa, 2015-10-30:
waypoints_navに以前実装したmove_baseの失敗検出処理がうまく働くことにより、自律移動は継続可能であることがわかったため閉じます。

closed 2015-10-30 by DaikiMaekawa