open-rdc/orne_navigation #125: orne_bringupが立ち上がらない

Nasupl, 2015-11-03:
最新で以下のエラーが発生してORNE βのorne_bringupが立ち上がりません
```
orne-beta@ornebeta-ThinkPad-X240:~/catkin_ws$ roslaunch orne_bringup orne_beta.launch 
... logging to /home/orne-beta/.ros/log/29da1202-8229-11e5-838d-28b2bd2b1498/roslaunch-ornebeta-ThinkPad-X240-16878.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/orne-beta/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

Traceback (most recent call last):
  File "/opt/ros/indigo/share/xacro/xacro.py", line 60, in <module>
    xacro.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 696, in main
    eval_self_contained(doc)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 626, in eval_self_contained
    eval_all(doc.documentElement, macros, symbols)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 526, in eval_all
    (str(name), str(node.tagName)))
xacro.XacroException: Invalid parameter "size" while expanding macro "xacro:sensor_imu"
while processing /home/orne-beta/catkin_ws/src/icart_mini/icart_mini_driver/launch/icart_mini_drive.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/orne-beta/catkin_ws/src/orne_navigation/orne_description/urdf/orne_beta.xacro'] returned with code [1]. 

Param xml is <param command="$(arg model)" name="robot_description"/>
The traceback for the exception was written to the log file
```

DaikiMaekawa, 2015-11-03:
icart_miniリポジトリとorne_navigationの両方を最新版にしていますか?

DaikiMaekawa, 2015-11-03:
> xacro.XacroException: Invalid parameter "size" while expanding macro "xacro:sensor_imu"

このエラーはicart_miniのバージョンが一つ低いことによるものだと推測されます。

Nasupl, 2015-11-03:
@DaikiMaekawa 
立ち上がりました。すいません。

closed 2015-11-03 by Nasupl