open-rdc/orne_navigation #126: スタート/リスタート時にコストを消したい

Nasupl, 2015-11-03:
/syscommandで"start"や"restart"を送った時点でコストをクリアしてスムーズにスタートするようにしたほうが良い気がする

DaikiMaekawa, 2015-11-03:
@Nasupl 運用でカバーできませんか? どのような問題が起きているのか報告していただけると助かります。

Nasupl, 2015-11-03:
特にスタート時に、前のチームが直前に出走している場合、その前にplay_waypoints_navなどを立ち上げているとスタート地点がコストまみれでリカバリ動作をしてからスタートすることになり、動作が不安定になります。
スタートまでの時間が決められている以上、あまりスタート地点に長居したくないので、スタート地点についてからplay_waypoints_navを立ち上げたくはありません。
リスタート時においても同様です。看板が目の前に置かれるため、スタート時に経路上にコストが発生して大きく迂回して横断歩道に進行する形となり、あまりよくないです。

DaikiMaekawa, 2015-11-03:
なるほど。 去年はあまり気にしていなかった問題ですね。 それについては対処しておきます。

DaikiMaekawa, 2015-11-05:
実装しました。 動画確認をお願いします。
https://github.com/open-rdc/icart_mini/pull/102

Nasupl, 2015-11-06:
コールバックの中でサービスを呼ぶようにする

Nasupl, 2015-11-06:
実装した
https://github.com/open-rdc/icart_mini/pull/103

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/issues/126 にも実装されました。 こちらはストップポイントの取り扱いに関しても改善されているため優先してテストを行う。

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/pull/129 にてコストマップのクリア処理が問題なく行われることを確認しました。

closed 2015-11-07 by DaikiMaekawa