open-rdc/orne_navigation #128: waypoint間の経路が左側に偏り，歩道から出てしまう

iou16, 2015-11-04:
各waypoint間を最短で(直線的に)走行せず，左側に膨らんだような曲線の経路が生成される．その結果，歩道から公園の芝生側へ縁石を乗り越えてしまう．現状縁石を乗り越えて芝生の上を走行してもその後waypointに向かっていく過程で，コースに復帰するが，オドメトリの誤差等が蓄積する危険があるため，このような経路が生成されるのを避けたい．
実際のbagファイル：http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2015-11-04-15-21-59.bag

DaikiMaekawa, 2015-11-04:
bagファイルを見てみました。 一応問題発生時はその内容 + bagファイルでの時間を教えてください。 データから推測は出来ますが実物を見ているわけではないので問題発生箇所を誤解してしまうと見当違いのアドバイスをしてしまう可能性があります。

DaikiMaekawa, 2015-11-04:
指令値通りに機体が追従できず偏った経路を生成し続けることがあります。 (去年もありました)
問題が発生している場所の路面はどうなっていますか? トルクのパラメータを変更することで解決することもあります。

DaikiMaekawa, 2015-11-04:
あと昨日のbagファイルにも同様の現象が見られているのですが、ロボットの位置とは随分離れた位置にオドメトリが生成されているのが気になります。

[image]

iou16, 2015-11-04:
bagファイル開始から90秒付近でwaypoint間(Namespaces:waypoint3→waypoint4)のpathを確認して頂けると解ると思いますが，左に膨らんだpathが生成されています．(今回アップロードしたbagファイルでは，他のwaypoint間でもこのような経路が生成されている傾向があると思います)．このような経路が生成されると，歩道と公園の境目の縁石(高さ的に地図では見えませんが)に乗り上げる危険性がありこのことを問題として捉えています．

iou16, 2015-11-04:
路面については少し下り坂になっていますが，路面に土砂や落ち葉，どんぐり等が山積している訳ではありません．

iou16, 2015-11-04:
ナビゲーション開始時にオドメトリとた正しい位置にずれがあるのは，恐らくナビゲーション開始前にロボットを持ち上げて移動したからだと思います．走行中に横方向にスライドするような走行軌跡が発生しているのは，縁石に乗り上げていたのだと思います．

DaikiMaekawa, 2015-11-04:
なるほど、シミュレータでは見られない挙動なので少し気になっていました。 念の為次回は手で持ち運ばずにそのままスタートさせてみてオドメトリがどのように出力されるか見てみて下さい。 トピックを見る限りはおかしな値が出力されているわけではないので問題ないと思いますが。

DaikiMaekawa, 2015-11-04:
さて本題に入りますが、左に膨らんだパスが生成された時刻は具体的には100sec以降くらいですかね。 
[image]

パーティクルの広がりすぎたケースを疑いましたがどうやら違うようです。 パスは自分の現在の位置から目標値までまっすぐ引かれていますが、機体が左側に逸れる度に修正されているためグローバルプランナーの問題では無いと思います。 確かにホイールオドメトリの結果を可視化するとウェイポイント間で円弧を描いているように見えます。 そしてローカルプランナーからは機体の傾きを修正するための制御指令値が出力されています。

左側に逸れるというと以下の問題を思い出しました。 本件にも関係あるかもしれませんし、無いかもしれません。もう少し時間が取れる時に詳しく解析してみます。
https://github.com/open-rdc/orne_navigation/issues/53

DaikiMaekawa, 2015-11-04:
> 走行中に横方向にスライドするような走行軌跡が発生しているのは，縁石に乗り上げていたのだと思います．

左側にズレる問題が発生しているのはグローバルパスの計画時なのか、それとも追従に失敗しているのかを明確にする必要がありますが、実物の挙動を見れないので何とも言えないですね。 bagファイル上ではプランナーに問題があるようには見えませんでした。 しかし、どのタイミングで縁石に乗り上げているのか等はこちらでは正確な判断が出来ません。

DaikiMaekawa, 2015-11-04:
@iou16 問題の発生した機体で単純に直進の指令をトピックで与えてみて直進性が確保されているかどうか確認をお願いします。 (navigationは起動しないでください)

iou16, 2015-11-04:
@DaikiMaekawa 単純に並進方向のcmd_velをトピックとして与えた場合でも少しずつ左に旋回するのを確認しました．
一つの素朴な質問なのですが，move_base/NavfnROS/planというトピックはどのようなトピックなのでしょうか？

iou16, 2015-11-04:
[image]
[image]

DaikiMaekawa, 2015-11-04:
navfnによって最後に計算されたnav_msgs/Pathというメッセージ型のトピックです。 ROS側のNavigationが使用したパスが主に可視化を目的にパブリッシュされています。

DaikiMaekawa, 2015-11-04:
まずは問題の発生した機体でypspurに関するパラメータが適切に設定されているかを再確認する。 (トルクの設定等)

アルファ、ベータ間でパラメータの差異が存在しローカルで管理していると以前聞きました。 そのパラメータに関して私は把握していないので何とも言えませんが、問題の起きていない機体からファイルを取ってきて動作確認等もしてみたほうが良い。 それら全ての確認が終了し、ハードウェアトラブルでもないとすると今までにない新しい問題ということになります。 そうなった場合、パラメータ調整等で改善するかどうかの検討はそちらにお任せします。

Nasupl, 2015-11-05:
@DaikiMaekawa 
YPSpurのパラメータは現在alpha, betaともに同じです。
機体の直進性が確保されておらず、若干左に旋回しながら走行するのは、YPSpurのパラメータを現在のパラメータに変更する以前から発生していた問題でしたが、ローカルなパスプランで修正されるため、大きな問題とされてこなかったのだと思います。

https://github.com/open-rdc/orne_navigation/issues/128#issuecomment-153754892 で示されている、グローバルパスプランが左に弧を描く問題は、機体の直進性の問題とは別なように思えるのですが、そうではないのでしょうか。

DaikiMaekawa, 2015-11-05:
@Nasupl 一応私がYPSpur関係のパラメータを調整した際は完全にNavigationとは切り離していましたので、指定の速度で直進性が確保されるかどうかはローカルプランナによる補正抜きで十分確保されていたと記憶しています。 ただし、それは指定する速度に依存していましたので、linear.xにいくつを指定した時にどの程度旋回してしまうのかという情報が重要かもしれません。 私が調整した時点では0.5m/sで直進させた場合、円弧を描く挙動を示すことは無くなりました。

> #128 (comment) で示されている、グローバルパスプランが左に弧を描く問題は、機体の直進性の問題とは別なように思えるのですが、そうではないのでしょうか。

その画像には説明がなかったため直前のコメントにあるmove_base/NavfnROS/planの質問に関する補足だと理解していました。 確かに円弧を描いていますが、コースアウトする危険性がある程には見えません。 (少なくともbagファイル上は)
ログを見るとそのパスを追従出来ず左側に機体が逸れていき、それを修正するためのパスをプランニングするということが繰り返されていましたので機体の直進性の方がこの問題への影響度が高いと判断しました。

DaikiMaekawa, 2015-11-05:
ただし、縁石が画像に示されているプランニング後のパスの上に存在しロボットがそのパスを正確に追従しているにも関わらず縁石に乗り上げたことによって横方向にスライドするような挙動を示しているという話であれば私の指摘は的外れということになります。 これに関してはbagファイルからの情報だけでは判断しかねます。

DaikiMaekawa, 2015-11-05:
グローバルプランナーは単純に点の間を直線で結ぶ最短経路を生成するわけではありません。 探索空間にはロボットの姿勢も含まれており、パスの生成に影響します。 waypoints_navノードはウェイポイントへ移動するためにmove_base側へ要求している姿勢は必ず(0, 0, 0, 1)となっているためそれが影響している可能性があります。 我々が使用しているウェイポイントの定義はx, yのみですが姿勢も含めたものに拡張した方が良いかもしれないと今思いました。 しかし、大会も近いのでウェイポイントの設定にこれ以上労力を割きたくはないですので、手軽にやるならその間に更にウェイポイントを追加するとかが妥当な線ですね。

DaikiMaekawa, 2015-11-05:
そもそも機体が低速時でも直進しないという場合はそちらの解決を優先した方が良いとも思いますが。

Nasupl, 2015-11-05:
@DaikiMaekawa 
機体の直進性については，0.8m/sでしか検証していませんでしたので，現在 @iou16 が低速の場合について検証中です．
Waypointの数を増やす件については， @yasuohayashibara 先生からもご指摘をいただきましたので，実施します．

iou16, 2015-11-05:
@DaikiMaekawa 
機体の直進性を0.5m/sで検証しました
検証の条件は
* 2号館3階の廊下で検証
* 床のタイルの等を基準に位置合わせ
* キャスターの向きも確認
* rostopic pub /icart_mini/cmd_vel geometry_msgs/Twist -- '[0.5, 0.0, 0.0]' '[0.0, 0.0, 0.0]'

検証した結果，6m程度走らせてみましたが，安定して直進しています(6m走らせても横方向の誤差は1cm以内かと)．

DaikiMaekawa, 2015-11-05:
推測ですが、バッテリーの残量等で機体側の挙動が変わったりしませんかね。 左側に軌道が逸れる原因はYPSpur側とwaypoints_navigationノードによるものでそれぞれ独立した原因があると考えています。
状況を整理すると、

1. ウェイポイント到着後に次の目標へとプランニングした際に発生する僅かに円弧を描くパス
    * ウェイポイントに姿勢を定義していないことが影響しているのではないか

2. その後も経路を追従できず左側に機体が逸れ続け、パスプランナはそれを修正しようとする
    * 以前発生していた機体の直進性が関係しているのではないか

> ただし、縁石が画像に示されているプランニング後のパスの上に存在しロボットがそのパスを正確に追従しているにも関わらず縁石に乗り上げたことによって横方向にスライドするような挙動を示しているという話であれば私の指摘は的外れということになります。 これに関してはbagファイルからの情報だけでは判断しかねます。

そういえば、これについて回答してもらってもいいですか?

DaikiMaekawa, 2015-11-05:
@iou16 の検証により試験走行時に見られたような0.4m/s以上で直進させた際に左側に逸れる挙動が再現されない。 しかし、依然として問題２が解決されていないため、Issue冒頭にあるように

> 各waypoint間を最短で(直線的に)走行せず，左側に膨らんだような曲線の経路が生成される．その結果，歩道から公園の芝生側へ縁石を乗り越えてしまう．現状縁石を乗り越えて芝生の上を走行しても，その後waypointに向かっていく過程でコースに復帰するが，オドメトリの誤差等が蓄積する危険があるため，このような経路が生成されるのを避けたい．

コースアウト = 問題１のみとする考え方は出来ない。 つまり従来のYPSpur側の問題とは関係しない原因が存在するかもしれない。 大会へ向けてはウェイポイントの追加で凌げる可能性も高いが、システム上重大な問題が未解決状態になっていることは記憶に留めておく必要がある。

iou16, 2015-11-05:
> 推測ですが、バッテリーの残量等で機体側の挙動が変わったりしませんかね。

ジョイスティックでβを運用した際の個人的な印象ですが，バッテリーを消耗した状況では直進性が低下する印象があります．因みに，今回私が検証に使用したバッテリーは充電済みのものです．

> ただし、縁石が画像に示されているプランニング後のパスの上に存在しロボットがそのパスを正確に追従しているにも関わらず縁石に乗り上げたことによって横方向にスライドするような挙動を示しているという話であれば私の指摘は的外れということになります。 これに関してはbagファイルからの情報だけでは判断しかねます。

今回のbagファイルを記録した際の動画が存在しないため，はっきりとしたことが確認できないというのが正直なところです．申し訳ありません．私の記憶では，今回アップロードした画像はwaypoint3→wayopint4とwaypoint4→waypoint5の画像だったと思います．この走行の際，縁石に乗り上げ復帰したのは，waypoint2→waypoint3だったと記憶しております．なので，今回アップロードした画像のパスを追従している際は，横方向にスライドするような挙動はなかったと思います．

iou16, 2015-11-05:
@DaikiMaekawa 
> "左側に膨らんだような曲線の経路が生成される．" (問題1)
コースアウト = 問題１のみとする考え方は出来ない。 つまり従来のYPSpur側の問題とは関係しない原因が存在するかもしれない。 大会へ向けてはウェイポイントの追加で凌げる可能性も高いが、システム上重大な問題が未解決状態になっていることは記憶に留めておく必要がある。

コースアウトの原因が，問題1のみとするのには無理があるのは私も同意です．今回のbagファイルを記録した際の自律走行は，試走会の終盤だったので，バッテリーの残容量が低下していた可能性は十分にあります．また，問題1が発生しているのも事実だと思います．何故なら，waypoint通過直後に生成された次のwaypointへのパス自体が湾曲していますし，このような挙動はバッテリーが十分に充電されていた際の走行でも発生していたからです．
ただ，一つ気になることとして，このように左側に膨らんだ曲線のパスが生成されるのが，必ず同じこの道で，その他の場所では発生せず(ただ，この道以外の道の道幅が広いため，あまり問題になっていないだけかも知れませんが)，この道では，どのwaypoint間でもこのように左側に膨らんだような曲線の経路が生成されていることです．

DaikiMaekawa, 2015-11-05:
> 自律走行は，試走会の終盤だったので，バッテリーの残容量が低下していた可能性は十分にあります．

金曜日も長い時間作業すると思いますのでバッテリーの残量と直進性に関連性があるかどうかに関しては運用時に注目してみてください。 

> ただ，このように左側に膨らんだ曲線のパスが生成されるのが，必ず同じこの道で，その他の場所では発生せず(ただ，この道以外の道の道幅が広いため，あまり問題になっていない=コースアウトしないだけかもしれませんが)，この道では，どのwaypoint間でもこのように左側に膨らんだような曲線の経路が生成されているのが少し気になっております．

現在の姿勢と目標間の角度の差がだいたい１８０度だからじゃないですか?

DaikiMaekawa, 2015-11-06:
> 金曜日も長い時間作業すると思いますのでバッテリーの残量と直進性に関連性があるかどうかに関しては運用時に注目してみてください。

今日のミーティングの内容をまとめます。 バッテリーがフル充電の場合でも大会の環境下では左側に逸れる挙動が見られた。 ジョイスティックのみ、Navigationを条件から排除した状態でもバッテリーの残量が減るにつれて直進性が失われていくことを実機で確認した。 https://github.com/open-rdc/orne_navigation/issues/135 の対応だけでコースアウトの危険性を排除することが出来ないのはbagファイルによる事前の検証で判明しているため、暫定的な対処法としてはバッテリー残量をなるべくフル状態に保つことを意識するという運用面でカバーするしかなさそうだ。

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/pull/129 の修正によりウェイポイント間の経路が弧を描いている問題は改善された。 

https://github.com/open-rdc/orne_navigation/issues/147 のmove_baseの更新周期のエラーによるコースアウトとバッテリー残量低下に伴う直進性の低下という問題を抱えているがそれはまた別の問題である。

closed 2015-11-07 by DaikiMaekawa