open-rdc/orne_navigation #132: ゴール付近で自己位置推定がうまく行かずにゴールに入れない

Nasupl, 2015-11-06:
* 要bagファイル添付

Nasupl, 2015-11-06:
動画
https://youtu.be/UkhlSZ2UllM

Nasupl, 2015-11-06:
@ShotaroTerato 
bagファイルお願いします

DaikiMaekawa, 2015-11-06:
@iou16 そちらでバグファイルを元に検証してお手上げだったら私が引き継ぎます。 今日は他の問題も合わせて長い戦いになりそうですので、出来れば負担を軽減したいと考えています。

ShotaroTerato, 2015-11-06:
bagファイルになります
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2015-11-06-21-50-57.bag

ShotaroTerato, 2015-11-06:
#139 にて対応。
[image]

DaikiMaekawa, 2015-11-07:
@iou16 amclのパラメータ調整はいかがでしょうか？

iou16, 2015-11-07:
z_hitのパラメータを変更して対応しようとしようとしたのですが，そうすると，その他の環境で自己位置推定が不安定になり,困っています.

DaikiMaekawa, 2015-11-07:
開けた場所でパーティクルの分散が広がりすぎる問題は結局解消されなかったが、ウェイポイントの調整によりゴールの枠には間違いなく入るよう暫定的に対処した。 まっすぐ走行しない問題にはmove_baseの制御周期エラーなども影響しているためそちらの問題を解消することを優先して行う。 開けた場所に於ける自己位置推定のパラメータ改善は来年度の重要な課題である。

DaikiMaekawa, 2015-11-08:
開けた場所に於けるamclのパラメータの最適化に関する検討を2016年度に向けて再開する。

Nasupl, 2016-06-14:
#113 に引き継ぎ

closed 2016-06-14 by Nasupl