open-rdc/orne_navigation #136: ストップポイントで回転してしまう

Nasupl, 2015-11-06:
* move_baseで姿勢が収束しない問題
* bagファイルと動画を添付する

Nasupl, 2015-11-06:
@ShotaroTerato bagを頼みます
@Nasupl 動画を上げる

Nasupl, 2015-11-06:
動画
https://youtu.be/JTBLvJAKYXE

ShotaroTerato, 2015-11-06:
bagになります
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2015-11-06-21-30-17.bag

DaikiMaekawa, 2015-11-06:
動画を確認しました。 最終的な判断はbagファイルを見てから行うつもりですが、
 > move_baseで姿勢が収束しない問題

という言葉から受けるイメージとは異なる現象が発生していると感じました。

DaikiMaekawa, 2015-11-06:
[image]

bagファイルからmove_baseへと送られたゴールを青色の矢印で可視化してみました。 ロボットが回転している方向とまったく同じですのでこちらは停止する際の姿勢を定義することで対処出来そうです。 収束しないと聞いて随分昔に修正された問題ですが目標角度に到達せずに右旋回 -> 行き過ぎて -> 左旋回 以下ループするといった現象を思い出しました。 なので念の為bagファイルを送って頂きました。

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/pull/129 にて、suspend_poseとして停止位置の姿勢も定義出来るようになった。 この姿勢を横断歩道に合わせて設定することによりこの問題は解決すると思われる。

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/pull/129 停止位置に姿勢を定義し、実機で問題なく動作することを確認した。

closed 2015-11-07 by DaikiMaekawa