open-rdc/orne_navigation #140: IMUのデータ受け取りが失敗する

ximo112, 2015-11-06:
自律走行中にIMUからのデータの受け取りが失敗し、pathが生成されなくなり壁に衝突する

ximo112, 2015-11-06:
@iou16 bagと動画をおねがいします

DaikiMaekawa, 2015-11-07:
先に案だけ書いておきます。 通信切断時にimuのトピックを投げないようにcit_adis_imuを修正すればrobot_pose_ekf側の各データに対するタイムアウトが働いてホイールオドメトリのみを使用してくれます。 そちらで手が空いていれば実装お願いします。 私が修正する場合は今のタスクが完全に終了した後になってしまいます。

iou16, 2015-11-07:
bagファイルをhttp://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2015-11-06-22-10-04.bag にアップロード致しました．

ximo112, 2015-11-07:
動画
https://www.youtube.com/watch?v=0bpzysiV_ik

Nasupl, 2015-11-07:
@DaikiMaekawa 
IMUのノードは、接続遮断時にはトピックを配信していませんし、チェックサムエラーが発生した場合も、その段階で例外が投げられてトピックを配信するところまで行かないので、問題発生時にはトピックは配信されないはずです。

DaikiMaekawa, 2015-11-07:
ふーむ、これからbagファイルも見てみて対策を考えてみます。

DaikiMaekawa, 2015-11-07:
> IMUのノードは、接続遮断時にはトピックを配信していませんし、チェックサムエラーが発生した場合も、その段階で例外が投げられてトピックを配信するところまで行かないので、問題発生時にはトピックは配信されないはずです。

トピックを吐き続けてますね。

ximo112, 2015-11-07:
ほぼ同じ場所において同現状が起きます、、、

DaikiMaekawa, 2015-11-07:
@ximo112 再発時との共通点を探りたいのでbagファイルをお願いします。

DaikiMaekawa, 2015-11-07:
デバッグした結果、IMUは正常に動作していることがわかった。 コース後半のため、バッテリー残量が減ることによる直進性の低下(https://github.com/open-rdc/orne_navigation/issues/128)やmove_baseの制御周期に関する警告が出力されていることの影響が大きい。 これに関しては別のIssueを作成し、対処することとする。

closed 2015-11-07 by DaikiMaekawa