open-rdc/orne_navigation #147: move_base outputs the warning of planning rate.

DaikiMaekawa, 2015-11-07:


DaikiMaekawa, 2015-11-07:
EKFのtf更新周期が10Hzと他に比べて低いことが影響していると推測される。

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/blob/indigo-devel/orne_bringup/launch/includes/base.launch.xml#L4 を修正して警告が出なくなることを確認すること。

恐らく30Hzくらいが妥当。

iou16, 2015-11-07:
@DaikiMaekawa 
質問なのですが，読み込むmapファイルのサイズによって"，map update loop missed its desired rate of 1.000Hz... the loop actually took ◯◯ seconds"というwarnが出るかどうかは左右されるでしょうか？

DaikiMaekawa, 2015-11-07:
グローバルプランニングの探索空間が変わる場合は関係あるかもしれませんね。 その質問をするということはホテルで警告が再現出来ないということでしょうか? @iou16

DaikiMaekawa, 2015-11-07:
マップの大きさ、そして解像度はプランニングの処理の負荷に影響します。

DaikiMaekawa, 2015-11-07:
因みにメッセージを吐いている箇所がどこだか特定できますか? rqt_consolでノード名と発生箇所をデバッグしてみてください。

iou16, 2015-11-07:
node名はmove_baseです，発生箇所とはどういった意味でしょうか？

DaikiMaekawa, 2015-11-07:
メッセージをクリックすると警告を出しているプログラムの行数まで絞り込めます。

iou16, 2015-11-07:
[image]

DaikiMaekawa, 2015-11-07:
costmap2d内に問題の発生箇所があるようですね。 ざっとコードを確認してきますので少々お待ちを。

DaikiMaekawa, 2015-11-07:
@iou16 costmap_2dの解像度が一番効いてきそうです。 デフォルトは0.05meters/cellなので屋外用途では少し小さすぎたかもしれませんね。
以下にプルリクエストを送りましたので検証おねがいします。
https://github.com/open-rdc/orne_navigation/pull/148

iou16, 2015-11-07:
検証してみましたが，同じメッセージが出ます．

DaikiMaekawa, 2015-11-07:
解像度をかなり大きめに設定してみたら警告は無くなる? @iou16

iou16, 2015-11-07:
@DaikiMaekawa 解像度を0.5に変更して検証してみます．

iou16, 2015-11-07:
解像度をどれだけ落としても同じエラーメッセージが出ます．

iou16, 2015-11-07:
[image]

DaikiMaekawa, 2015-11-07:
https://github.com/open-rdc/orne_navigation/pull/148 にて対応完了。 
グローバルコストマップとローカルコストマップの解像度を下げて、EKFノードのtfの周期も上げた。
機体標準のPCではなく @Nasupl のPCを使用することで警告が表示されなくなり、CPUの使用率が下がることを確認。 

つまり、以下の環境で最低限の要求スペックを満たした。 
```
Core-I5, 10GB memory, HDD
```

DaikiMaekawa, 2015-11-08:
@Nasupl のPCは雨により破損、共有PCでは https://github.com/open-rdc/orne_navigation/issues/151 で挙げたように安定して完走することは困難。

Nasupl, 2016-06-14:
こちらの問題はRAMの不足が原因であると断定し，2016年度は各PCのRAM容量を増加することで解決としました．一旦閉じます．

closed 2016-06-14 by Nasupl