open-rdc/orne_navigation #151: 2015年度のつくばチャレンジ本走行で発覚した問題、改良点を洗い出す

DaikiMaekawa, 2015-11-08:
各々がここに思ったことを書き込んでいきましょう。

DaikiMaekawa, 2015-11-08:
### チームβのリタイヤの原因
話を聞いた限りだとナビゲーションシステムの制御周期の遅延により、パスが完全に生成される前の速度指令を使用して走行を続けてしまったためによりコースアウトしてしまったと推測されます。 

Q. なぜ遅延が発生するのか?
A. 大きく分けて三つの原因が存在します。

* 一つ目はmove_base内で処理されるtfの変換待ちの処理に関するものです。 EKFのノードからトランスフォームされるtfの周期は10Hzと他のフレームより低く、tfは分散システムでも変換が破綻しないようにタイムスタンプを厳密に取り扱います。 move_baseが要求するtfのレートは少し高めで、全体のパフォーマンスは10Hzのodomフレームに引っ張られてしまいます。

* 二つ目はPCの処理能力不足です。
コストマップの解像度を下げる、PCのメモリを4GBから10GBにした際にmove_baseから出力される制御ループの遅延に関する警告が出力されなくなり、機体の挙動が大幅に改善されます。 
amclは機体がまったく動いていない状態で1GBのメモリを消費、コース上での自律走行時にはパーティクルの数を適応的に増減させるため更に処理能力が必要となる。 play_waypoints_nav.launchを立ち上げるだけでも共有PCではCPUの70％を使用、時々100％に達する。 move_baseが必要としているPCの処理スペックのリファレンスと比較してもその水準には達していない。 この問題は走行速度を下げて、更にパラメータをフルチューニングし直すことで辛うじて解決することは可能だと推測されるが、現行の構成でポイントクラウドの処理などをリアルタイムで行い人物検出等も行うことは不可能。

2014度の本走行で使用した私のマシンのスペックを参考までに挙げておきます。
Core-I7, memory 8GB, SSD

* 三つ目の原因はPCの放熱です。 
なぜコース終盤になるほどワーニングの頻度が上がり、機体の挙動も不安定になっていくのでしょうか? この疑問への回答は単純です。
この問題は二つ目の問題から派生したものなのですが、ただでさえCPUを100％付近で使用しているPCの温度が上昇することによって発生する処理能力低下が自律走行を継続するにあたって致命的なレベルに達しています。

### チームαのリタイヤの原因

* タイヤ半径が小さすぎる。
道に存在する小さな溝にハマり車輪がスタックし、そのままリタイヤとなりました。
リアル・ワールドではこのような整備が行き届いていない箇所がいくつも存在します。 この問題に関してはハードウェア的に解決するのが正解だと思います。 車輪径を大きくしたり、キャスターのバネを変更したりと改良点はいくつも存在します。

DaikiMaekawa, 2015-11-08:
本番のコースアウト以外で気になったことを挙げます。

* LRFが雨によって距離を誤検出することにより、コストマップが常にロボットの周囲に障害物を登録し結果として、雨が強くなるにつれて回避行動を繰り返す頻度が増加する。

LRFからのデータをフィルタリングしたり、コストマップの解像度を荒くすることで対処は出来ないか。

* ゴール付近等の開けていてランドマークの少ない領域ではパーティクルが散らばりすぎて走行が不安定になる。

amclのパラメータ調整は必須。

* 緩やかな傾斜では機体がピッチ方向に揺れた際に地面を障害物と誤検出することがある。 これがウェイポイントと重なることで誤検出した障害物を回りこむようなパスが生成される。

トライアルエリア等の開けたコースでマップ作成、自己位置推定の観点からなるべくランドマークを増やすためにURGは低めの位置に取り付けられている。 (2014年度に絶妙な高さに調整した)
しかし、障害物回避においては低めのURGは仇となるケースが多いため高い位置にもう一つ障害物回避用URGを追加してもいいかもしれない。 

または、3次元での姿勢をもう少し精度良く出せるようにして地面を誤検出しているかどうかの判定を加えることも可能だが、この方法だと角度の閾値のパラメータ等の調整がシビアになる。

DaikiMaekawa, 2015-11-08:
*  以前 https://github.com/open-rdc/orne_navigation/issues/97 で追加した障害物回避時の減速、及びリカバリー時の動作用に追加していた処理がリアル・ワールドでは有効に機能しなかった。

減速しすぎて車輪がスタックするなどの現象が起きている。 必要に応じて速度に関するパラメータの調整、もしくは新規にプラグインを開発することで対応する。

* バッテリー低下に伴って機体の直進性が低下する。 https://github.com/open-rdc/orne_navigation/issues/128 により判明。

https://github.com/open-rdc/orne_navigation/issues/53 にて2014年度はYPSpurのパラメータを調整して左側に逸れる挙動を改善していたが、バッテリー低下によって再発することが確認された。 本来i-Cart miniが使用しているモータとモータドライバは大出力に対応していてこれはパラメータを最適化することにより性能を引き出すことが出来る。 左に大きく逸れる問題に関してもYPSpur側が計算に使用しているパラメータを修正することで対応可能であると推測させる。

DaikiMaekawa, 2015-11-08:
各問題に対してIssueを作成し、担当者を再設定し直した後このIssueを閉じる予定。

yasuohayashibara, 2015-11-09:
パソコンの処理性能の話がありましたので，少し調べてみました．

■Lenovo ThinkPad X240 20AL006P JP
Core i5-4300U
ベンチマーク　3755
http://ark.intel.com/ja/products/76308/Intel-Core-i5-4300U-Processor-3M-Cache-up-to-2_90-GHz

■最新レッツノート　ハイエンド
Intel Core i7-6600U
ベンチマーク　4603
http://ark.intel.com/ja/products/88192/Intel-Core-i7-6600U-Processor-4M-Cache-up-to-3_40-GHz

ベンチマークの比較
http://www.cpubenchmark.net/compare.php?cmp[]=2608&cmp[]=2054

確かに差はありますが，それほど違いはなさそうな気がします．大量のデータを扱いますので，メモリの影響が大きいのかもしれません．
あとは，運用は難しくなりますが，ノートではなくBRIXのようにデスクトップ用のCPUを搭載するということになりそうです．

■BRIX　GB-BXI7-4770R
Intel Core i7-4770R
ベンチマーク　9817
http://www.cpubenchmark.net/cpu.php?cpu=Intel+Core+i7-4770R+%40+3.20GHz&id=2137
※上記は例えです．探せばもっと処理性能の高いものがあるかもしれません．

DaikiMaekawa, 2015-11-09:
> 確かに差はありますが，それほど違いはなさそうな気がします．大量のデータを扱いますので，メモリの影響が大きいのかもしれません．
あとは，運用は難しくなりますが，ノートではなくBRIXのようにデスクトップ用のCPUを搭載するということになりそうです．

BRIXは残念ながらオススメできませんね。 特につくばチャレンジ用途では。
メモリの大きさが重要という意見には同意します。 ８GB以上は必須かなというのが2014年度で運用していた者としての感想です。 ただし、あまりギリギリを攻めすぎるとポイントクラウドを取り扱う際や後々の機能の拡張に支障をきたしかねません。 消耗品として購入するため価格を抑えたいところだと思いますが、ここは有意義な投資ということで備品で十分なスペックを持つPCの購入を検討してみてはいかがでしょうか。

DaikiMaekawa, 2015-11-09:
以下のリンクでインテル社のモバイルパソコン向け CPUの性能比較表を見ることが出来ます。
http://pcinformation.info/cpu/comparison-intel-mobile.html

極端な例ではありますが、 @yasuohayashibara が挙げているレッツノートのCPUよりも２倍以上の性能を持つものも存在しますので、参考までに。

DaikiMaekawa, 2015-11-10:
> 一つ目はmove_base内で処理されるtfの変換待ちの処理に関するものです。 EKFのノードからトランスフォームされるtfの周期は10Hzと他のフレームより低く、またtfは分散システムでも変換が破綻しないようにタイムスタンプを厳密に取り扱います。 move_baseが要求するtfのレートは少し高めであるにも関わらず、全体のパフォーマンスは10Hzのodomフレームに引っ張られてしまいます。

https://github.com/open-rdc/orne_navigation/blob/indigo-devel/orne_bringup/launch/includes/base.launch.xml#L4 
30Hzに引き上げました。 現時点では実機での検証が不足しています。

DaikiMaekawa, 2015-11-10:
ウェイポイントの設定が職人芸。 もっと賢く環境情報をプランニングに反映させることを検討するのはどうか。 私からの案は４つでどれも3D-URGからのポイントクラウドデータを使用する。 別に一つしか選べないわけではないので必要に応じて組み合わせる。

1. コストマップの表現にElevation Mapを使用する。 
更に、各種プランナーのプラグインを高さ情報を使用できるように拡張する。 実装の手軽さと計算コストの低さが魅力。

2. OGMで表現されたコストマップの更新にポイントクラウドを活用する。
最近、costmap2dの対応リソースにポイントクラウドが追加された。 パラメータを調整するだけで我々の要求仕様を満たすようであれば新規に開発を行う必要がない。

3. 地図作成段階からElevation MapもしくはOctree等の使用を検討する。
3D-URGの計測範囲がそれほど広くないのがネックになると思われる。 また、Octreeはツリー構造であるが故に実装コストは他に比べると高いので注意。

4. 走行可能領域を編集可能なエディタを作成する。
この時高さ方向に関する情報を持つ地図表現を使用していた場合にはその情報を用いて効率的な編集が可能。 現実と認識結果の相違は少なからず発生するものなので走行可能な領域をGUI上でマーキング、自律走行時はその領域内のみを各種プランナーは使用する。

DaikiMaekawa, 2015-11-10:
我々のウェイポイントの定義は位置のみだが姿勢を含めるのはどうか。 (注) 現状は、次のウェイポイントの方向という常に固定の姿勢が定義されているとも解釈出来る。
任意の姿勢を含める方がより一般的且つ設定に柔軟性がある。 (その柔軟性が我々のシステムに必要かどうかは要議論)
ROS上の都合で言えばgeometry_msgs/PoseArrayに全てのデータを格納でき、可視化コードやyamlのロード・セーブのコードを全て削ることが可能。 ゴール時の姿勢を別途定義する必要も無くなるので実装がよりシンプル且つ簡単になる。

Hiroki-Goto, 2015-11-12:
昨日からORNE-βのパソコンのキーがずっと押されている状態になっているのでログインができなくなりました。
パスワード入力の画面で起きているのでどのキーが入力されているかは分かりません。

Nasupl, 2015-11-15:
[image]

Nasupl, 2017-05-30:
とりあえず閉じます

closed 2017-05-30 by Nasupl