open-rdc/orne_navigation #162: udev rule が適用されない

Hiroki-Goto, 2015-12-05:
PCにセンサなどをさしたときにUSBがさされたことは認識されますがudev ruleが適用されなくなりました

Hiroki-Goto, 2015-12-05:
IMUのruleが適用されていないPC（URGは認識されている）にIMUのruleが適用されているマシンのIMUに関するruleのみをコピーしたところIMUは認識されるようになったがURGが認識されなくなった

Nasupl, 2015-12-07:
双方のruleをここに貼り付けてもらえると助かります

Nasupl, 2015-12-08:
αのルールです
SUBSYSTEMS=="usb", KERNEL=="ttyACM[0-9]*", ACTION=="add", ATTRS{idVendor}=="15d1", ATTRS{idProduct}=="0000", MODE="666", PROGRAM="/opt/ros/indigo/env.sh rosrun hokyuo_node getID %N q", SYMLINK+="sensors/hokuyo_%c", GROUP="dialout"
SUBSYSTEMS=="usb", KERNEL=="ttyACM*", ACTION=="add", ATTRS{product}=="T-frog Driver", MODE="666", SYMLINK+="sensors/icart-mini", GROUP="dialout"
SUBSYSTEMS=="usb", KERNEL=="ttyUSB*", ACTION=="add", ATTRS{product}=="FT232R USB UART", MODE="666", SYMLINK+="sensors/ftdi%n", GROUP="dialout"

Nasupl, 2015-12-08:
βのルールです
SUBSYSTEMS=="usb", KERNEL=="ttyACM[0-9]*", ACTION=="add", ATTRS{idVendor}=="15d1", ATTRS{idProduct}=="0000", MODE="666", PROGRAM="/opt/ros/indigo/env.sh rosrun hokyuo_node getID %N q", SYMLINK+="sensors/hokuyo_%c", GROUP="dialout"
SUBSYSTEMS=="usb", KERNEL=="ttyACM*", ACTION=="add", ATTRS{product}=="T-frog Driver", MODE="666", SYMLINK+="sensors/icart-mini", GROUP="dialout"
SUBSYSTEMS=="usb", KERNEL=="ttyUSB*", ACTION=="add", ATTRS{product}=="FT232R USB UART", MODE="666", SYMLINK+="sensors/ftdi%n", GROUP="dial

Nasupl, 2015-12-08:
```
PROGRAM="/opt/ros/indigo/env.sh rosrun hokyuo_node getID %N q", SYMLINK+="sensors/hokuyo_%c", 
```
が適用されない原因っぽい。こいつを消してデバイス名をhokuyoにすれば問題なく動く。

Hiroki-Goto, 2015-12-08:
これで自律移動できるようになりました。
ありがとうございます

closed 2016-01-25 by DaikiMaekawa