open-rdc/orne_navigation #164: record_waypointが起動しない

Hiroki-Goto, 2016-02-10:
record_waypoint を起動したときrvizは起動するがwaypointを置くことができない（初期位置の設定だけは可能）

Nasupl, 2016-02-14:
見るからに起動時にエラーメッセージ出てるんだから貼っておいてほしかったですね
```sh
icart-mini-x240@icartminix240-ThinkPad-X240:~$ roslaunch orne_navigation_executor record_waypoints_joy.launch 
... logging to /home/icart-mini-x240/.ros/log/50ef2d4e-d2bc-11e5-a300-f55e875fe091/roslaunch-icartminix240-ThinkPad-X240-2974.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/icart-mini-x240/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://icartminix240-ThinkPad-X240:50236/

SUMMARY
========

PARAMETERS
 * /amcl/gui_publish_rate: 10.0
 * /amcl/initial_cov_aa: 0.0
 * /amcl/initial_cov_xx: 0.0
 * /amcl/initial_cov_yy: 0.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 0.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.05
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 30
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 5000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.2
 * /amcl/odom_alpha2: 0.2
 * /amcl/odom_alpha3: 0.8
 * /amcl/odom_alpha4: 0.1
 * /amcl/odom_alpha5: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.1
 * /amcl/update_min_a: 0.5
 * /amcl/update_min_d: 0.2
 * /joy_node/autorepeat_rate: 20
 * /joy_node/deadzone: 0.3
 * /joy_node/dev: /dev/input/js0
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /teleop_twist_joy/axis_angular: 0
 * /teleop_twist_joy/axis_linear: 1
 * /teleop_twist_joy/enable_button: 4
 * /teleop_twist_joy/enable_turbo_button: 5
 * /teleop_twist_joy/scale_angular: 0.9
 * /teleop_twist_joy/scale_linear: 0.6
 * /teleop_twist_joy/scale_linear_turbo: 0.8
 * /waypoints_saver/filename: /home/icart-mini-...
 * /waypoints_saver/save_joy_button: 0

NODES
  /
    amcl (amcl/amcl)
    joy_node (joy/joy_node)
    map_server (map_server/map_server)
    rviz (rviz/rviz)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
[... 39 lines cut ...]
ROS path [30]=/home/icart-mini-x240/catkin_ws/src/fulanghua_navigation/fulanghua_ekf_2d
ROS path [31]=/home/icart-mini-x240/catkin_ws/src/fulanghua_navigation/fulanghua_nav_demo
ROS path [32]=/home/icart-mini-x240/catkin_ws/src/fulanghua_navigation/fulanghua_srvs
ROS path [33]=/home/icart-mini-x240/catkin_ws/src/orne_navigation/orne_rviz_plugins
ROS path [34]=/home/icart-mini-x240/catkin_ws/src/fulanghua_navigation/fulanghua_waypoints_nav
ROS path [35]=/home/icart-mini-x240/catkin_ws/src/orne_navigation/orne_strategy
ROS path [36]=/opt/ros/indigo/share
ROS path [37]=/opt/ros/indigo/stacks
process[amcl-5]: started with pid [3021]
process[rviz-6]: started with pid [3046]
[ INFO] [1455414225.506467982]: Requesting the map...
[ WARN] [1455414225.507144706]: Request for map failed; trying again...
[ INFO] [1455414226.050966302]: Received a 4000 X 4000 map @ 0.050 m/pix

[ INFO] [1455414226.291284056]: Initializing likelihood field model; this can take some time on large maps...
[ INFO] [1455414226.530490185]: Done initializing likelihood field model.
^C[rviz-6] killing on exit
[amcl-5] killing on exit
[map_server-3] killing on exit
[teleop_twist_joy-2] killing on exit
[joy_node-1] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```

Nasupl, 2016-02-14:
つくばチャレンジ以降、旧waypoint_navで作ったwaypoint使ってたから気づかなかったけど、record_waypoints_joyで立ち上げるwaypoints_saverをfulanghua版に変えるの忘れてたっぽいですね

DaikiMaekawa, 2016-02-15:
@Hiroki-Goto 使用環境とエラーログを報告する癖は付けておいた方が良いと思います。 これは他のオープンソースプロジェクト等でも共通すると思いますが、より迅速に正確な回答をもらえる可能性が高くなります。

commit 1fa0a9fa (2016-02-15): Merge pull request #165 from open-rdc/fix_#164

fix #164
commit cbb4ad17 (2016-02-15): fix #164
closed 2016-02-15 by DaikiMaekawa