open-rdc/orne_navigation #169: icart_mini.xacroが実際のorneと異なった構成になっている

iou16, 2016-04-17:
icart_mini.xacro上での車輪の位置が，シミュレータ上での挙動を自然なものにするために，実際の位置と異なっており，本来存在しない受動輪も存在する．
これは実際のロボットの挙動にも影響を与える(RoboCup@Homeでは問題となった)為，実際の配置に合わせるべきである．
もし，シュミレーター上で，これまでの挙動を維持する場合はicart_mini.xacroを別途用意すべきであり,その辺りは用検討．

iou16, 2016-04-17:
また，対処についてもorne_description側に新たにicart_mini.xacroを作成し，そちらを参照するようにするか，本家のicartをforkして修正しpull requestを送るかなどの対応の仕方も用検討．

iou16, 2016-04-17:
@DaikiMaekawa 
ご意見を頂けると幸いです．

DaikiMaekawa, 2016-04-18:
Homeだと問題になったのですね。 それならば、つくばチャレンジでも問題が表面化していないだけだと思います。
せっかくの機会ですので修正をorne_navigationだけに留めてしまうのは勿体ないのでは? icart_mini側にプルリクエストを投げるのが良いと思います。

iou16, 2016-04-19:
上記の問題に対しては https://github.com/open-rdc/icart/pull/115 を作成し対処しました．また，icart_mini.xacroを変更したことによってorne_alpha.xacorとorne_beta.xacroおよびcostmap_common_params.yamlに変更が必要となったので https://github.com/open-rdc/orne_navigation/pull/170 を作成し対応しました．実機での挙動は確認済みです．

iou16, 2016-04-22:
https://github.com/open-rdc/icart/pull/115 と#170 のmergeし，問題に対処することができたのでこのissueを閉じます．

closed 2016-04-22 by iou16