open-rdc/orne_navigation #186: 第一回実験走行会メモ

Nasupl, 2016-07-09:
### αの地図
* 昨年度地図とbridge1, before_cross, before_cross_2, after_crossを合成する必要がある
* 横断歩道と、中央公園内のランドマーク皆無エリアがうまく地図を作れない

### 忘れもの・買い忘れ
* モバイルバッテリーからPCに給電するためのDCプラグがついたケーブル
* Aオス-Aメスのケーブル（Aオス-AオスケーブルとAオス-Bオスケーブルを誤って購入していた）

### 電源
* X240の大容量バッテリーのラッチが破損して使えなくなった（衝撃でPCから脱落し電源が落ちる）。テープで無理やり固定すれば使用できるかもしれない。
* E460がめちゃくちゃ電源を食う（大清水公園を一周するだけで20%くらい消費する）ので対策が必要
* βはモータを駆動しながら3DURGを動作させるような電力消費をしない使い方（地図をつくる、Waypointsを設定するなど）ならば、従来のバッテリーで十分かもしれない。

その他気づいた点があれば各位書き込むこと

Hiroki-Goto, 2016-07-09:
**βの地図**
トライアル区間と探索エリアを作った（トライアル区間は最後の方で角度がずれていったが画像を加工すればどうにかなると思う）
探索エリアの地図はとじ込み等うまくいかなかっため作り直す必要があるかもしれない

**２D－URGのudevルールが適用されない問題**
これは頻繁に起こる問題。いつもは70-sensor-ruleのURGのIDを取得する部分を消すことによって対応している。（今日もそれで対応した）

**αの電装系について**
車検時にテープなどで固定しておくようにと言われた

yasuohayashibara, 2016-07-10:
地図を作成したら，どこか(WebDAV?)にアップしてもらえればと思います．

みなさんから指摘があったように，ランドマークの少ない公園内で地図の作成が難しい．
また，自律走行も難しいので，ランドマークを増やす努力をすべきではないかと思います．
ランドマークとしては，以下が挙げられる思います．

１）地面の種類（アスファルトと草むら）
　　分類せずに反射強度をそのまま尤度の計算に使っても良いと思います．
２）２．５Dの地形データ
　　URGが取り付けられた水平面だけでなく，高さ方向で輪切りにした地形データ

他チームで既に導入されて有効性が確認されたものですが，安定性を向上させるためには本チームも導入したほうが良いのではと思います．

ShotaroTerato, 2016-07-11:
＊外装
icart-middleは高さ10~30cmの間に外装を追加する必要があり

buzzshima, 2016-07-12:
icart-middleの電源のたっぱのゴムブッシュの位置変更(?)

Nasupl, 2016-07-12:
- [ ] 報告書書く

closed 2016-08-09 by Nasupl