open-rdc/orne_navigation #189: 結界の実装

Nasupl, 2016-07-12:
走行禁止箇所に結界を設定できるようにしてほしい

ShotaroTerato, 2016-07-19:
amclとcostmap2dはそれぞれmapを受け取っている．
amclは今までどおりのmapを受け取り，costmap2d側で編集済みmapを受け取らせる方法が一番簡単な対処法か

Nasupl, 2016-07-19:
launchファイルを書いてください

Nasupl, 2016-07-26:
memo: map_serverを2つ建てる

ShotaroTerato, 2016-07-29:
map_serverを２つ建てた
今度はcostmap側に別のmapを受け取らせる
[image]
[image]

ShotaroTerato, 2016-07-29:
navigationのstatic layerのプラグインを直書き換えしたところ
うまくグローバルコストマップとしてもうひとつのマップが読み込まれた
[image]
[image]

ShotaroTerato, 2016-07-29:
マップの占有してるとこをpoint cloudに変換して障害物として認識させるほうが早いかも

ShotaroTerato, 2016-07-29:
amclがちゃんと元マップを受け取れているか確認をする

ShotaroTerato, 2016-08-01:
global costmap として別のマップを受け取れた
amclもちゃんと動いてるっぽい
[image]

ShotaroTerato, 2016-08-01:
運用時にいちいち2つのマップファイルを指定するの面倒になりそう

ShotaroTerato, 2016-08-02:
argを使ってうまい具合に読み込ませるようにする

ShotaroTerato, 2016-08-02:
[image]
[image]

ShotaroTerato, 2016-08-02:
efc2090 でmap_file指定時に自動的にmap_file_for_costmapの指定がなされるようにした
運用法が少し変わりmap_fileを指定するときは

・従来
map_file:=/フルパス/map.yaml
・変更後
map_file:=/フルパス/map

となった．
この時，“mapファイルとmap_for_costmapファイルを同じディレクトリに保存しておくこと”

Nasupl, 2016-08-09:
地図を用意すれば使える

Nasupl, 2016-08-31:
とりあえず検討は完了したので一回閉じる

closed 2016-08-31 by Nasupl