open-rdc/orne_navigation #200: 膨張リセットAMCLの適用

Nasupl, 2016-10-04:
@Shiina-Shiotani が開発した膨張リセット法を適用したAMCL(以下ER-AMCL)をorne_navigationに適用する

- [ ] Navigationパッケージをopen-rdcにフォークしてER-AMCLをコミットする
- [ ] orne_navigationの依存関係をフォーク側にする
- [ ] 動作確認

Shiina-Shiotani, 2016-10-11:
オーナー権限が無いとフォークできないみたいです
持っている方はフォークをお願いします

Nasupl, 2016-10-12:
@Shiina-Shiotani https://github.com/open-rdc/navigation へのpush等も権限がありませんか？

Shiina-Shiotani, 2016-10-13:
push できなかったので、ないと思います。

その時のエラーメッセージは以下の様になりました

remote: Permission to open-rdc/navigation.git denied to Shiina-Shiotani.
fatal: unable to access 'https://github.com/open-rdc/navigation.git/': The requested URL returned error: 403

yasuohayashibara, 2016-10-14:
open-rdcの権限を見るとownerは前川さん，後藤くん，林原のようです．
権限の問題であれば，後藤くんが対処するか，もしくは別の人にもowner権限を与えて対処できるようにしたほうが良いと思います．
夏迫くんがowner権限を持っていないのは不自然かなと思いますので，ownerに変更したいと思いますがよろしいでしょうか．

Nasupl, 2016-10-14:
@yasuohayashibara 是非お願いします

yasuohayashibara, 2016-10-14:
@Nasupl ownerに変更しました．

closed 2016-11-15 by Nasupl