open-rdc/orne_navigation #202: 機体にスポンサーのシールを貼る

Nasupl, 2016-10-11:
本走行までに貼る

NamikiHiroaki, 2016-10-24:
シール用紙検収済みです

Nasupl, 2016-10-25:
高解像度なロゴのデータがないのでそれ待ち

yasuohayashibara, 2016-10-26:
一応ロボカップで問合せてみましたが反応がないおそれがあります．
もし間に合わないようでしたら，ウェブ上のものを利用して下さい．

あと，フォーラムエンジニアリングのロゴだけだと「フォーラムエンジニアリング」チームになってしまいますので，千葉工業大学（もしくはORNE？）のロゴは最低限必要だと思います．

サイズや見え方に注意しないとみなさん「フォーラムエンジニアリング」の社員と間違われることになります．

ちなみに，未来ロボティクスのロゴを見つけました．（使わなくても良いが．．．）
[image]

yasuohayashibara, 2016-10-26:
高画質のロゴ手に入れました．

https://drive.google.com/file/d/0B1R8y8aFBfUdSnNJMmNyZERPb1E/view?usp=sharing
https://drive.google.com/file/d/0B1R8y8aFBfUdeE4tQjVaU0JZWFk/view?usp=sharing
https://drive.google.com/file/d/0B1R8y8aFBfUdVFR5ODhPQkUzbWM/view?usp=sharing

Nasupl, 2016-10-31:
貼った

closed 2016-10-31 by Nasupl