open-rdc/orne_navigation #209: 不足している備品の補充

Shiina-Shiotani, 2016-10-14:


Shiina-Shiotani, 2016-10-14:
10/15, 16 試走会前でのチェック
・ビニール袋
・マジックテープ
・両面テープ
・雑巾(予報では晴れなので必要ないかも)

buzzshima, 2016-10-18:
マジックテープ大量

ShunShimamura, 2016-10-18:
- [ ] マジックテープ

- [ ] ゴミ袋、ビニール袋

- [ ] タオルor雑巾

- [ ] 両面テープ

NamikiHiroaki, 2016-10-21:
ポリ袋（65cm×80cm 0.012mm 45L）40枚入り
雑巾（20cm×30cm）3枚入り
両面テープ（15mm×6.5m）
各１つずつ購入
マジックテープについては新習志野・津田沼周辺にはロール状のものがないのでネット購入になるかと

Nasupl, 2016-10-21:
@NamikiHiroaki マジックテープ，南船橋のスーパービバホームか船橋競馬場のコーナンまでいけばロールのやつ売ってるはず．

NamikiHiroaki, 2016-10-21:
Vivaホームの方は今日確認したのでコーナンの方寄ってみます

Nasupl, 2016-10-21:
あー，ロールというか切り売りのやつでいいと思う． @buzzshima, @ShunShimamura 両面テープ何m必要ですか？具体的に．

buzzshima, 2016-10-21:
最低5㍍ですかね

欲を言えば切り売りだと小さな余分なやつをなくしてすぐストックなくなりそう

buzzshima, 2016-10-21:
あと#202のシール用紙もお願いします

Nasupl, 2016-10-21:
切り売りって店頭にロールで置いてあって店員さんに切ってもらって買うタイプのやつのことですよ．
@buzzshima が言ってる切り売りは20cmごとにパッケージに封入されているやつのことでは？

buzzshima, 2016-10-21:
@nasupl  予想してるとおりです

Nasupl, 2016-10-21:
@NamikiHiroaki ということで1m辺りx円みたいな感じで売ってるロール切り売りのマジックテープを5mお願いします

NamikiHiroaki, 2016-10-21:
マジックテープの幅は15mmとかそのくらいで大丈夫でしょうか？

buzzshima, 2016-10-21:
なので@nasupl の言ったとおりで

Nasupl, 2016-10-21:
15mm細くない？25mmがベストだけどなければ15mmでいいです．

buzzshima, 2016-10-21:
@NamikiHiroaki 20ミリ以内であればできるだけ太めで割りにいいやつで

NamikiHiroaki, 2016-10-21:
了解しました

buzzshima, 2016-10-21:
25mmだとフレームからでるので少し嫌ですね

NamikiHiroaki, 2016-10-23:
切り売りは幅25mmが最小のようです

Nasupl, 2016-10-23:
@NamikiHiroaki それでいいです

NamikiHiroaki, 2016-10-24:
マジックテープ5m 検収済みです

Nasupl, 2016-10-24:
全部揃った

closed 2016-10-24 by Nasupl