open-rdc/orne_navigation #238: 経路生成手法の刷新

Nasupl, 2016-11-16:
### 現在の手法: 
1. 人間 (Waypoint(以下WP)を配置することによる大域的な経路生成)
2. global_planner （WP間のローカルな経路生成）（ローカル・グローバル双方のコストマップに影響を受ける）
3. base_local_planner (ローカルな経路に従う速度制御）
* この手法では，一つ一つのWPがゴールになっているため，WPの直上に障害物が存在すると到達できずに経路生成に失敗することがある．
* また，経路生成をWP毎に行うために遅い

### 必要とされている手法
1. 人間 (WPの配置による大まかな通過点の指定）
2. 大域的な経路生成器 (ローカルコストマップに影響を受けず，グローバルコストマップのみ基づいて全てのWPを最短で通過するパスの生成）
3. 局所的な経路生成器 （ローカルコストマップに基づいて，大域的な経路を追従する経路を生成する）
4. 速度制御器 （局所的な経路を追従するための速度制御）

ShotaroTerato, 2017-02-28:
デバッグ中
　WPをyamlから読み込む　済
　各WPごとの経路を計算　デバッグ中
　各経路を結合　デバッグ中

ShotaroTerato, 2017-02-28:
[image]
つながった経路の配信

しかし、local planner は動かず

ShotaroTerato, 2017-02-28:
WPをyamlから読み込む　済
各WPごとの経路を計算　済
各経路を結合　済

local plannerに反映させる New!

ShotaroTerato, 2017-03-06:
global plannerで繋げた経路をlocal plannerで追従するようにはできたが、経路上に障害物が現れるとlocal planner(dwa)で障害物回避してくれない。
dwaのパラメータの問題か

Hiroki-Goto, 2017-12-05:
neo navigation がダメだったらもう一回開く

closed 2017-12-05 by Hiroki-Goto