open-rdc/orne_navigation #239: 緊急脱出リカバリの@Homeからの輸入

Nasupl, 2016-11-16:
2016年度の本走行において，MCLのパーティクルが地図上で占有されているセルの中にめり込んだ上，そのパーティクルの尤度が最尤になったため，経路生成が正しく行えず身動きが取れなくなったことがあった．
このような状態では，ロボットが動ける範囲で動くことによって，自己位置の回復に努めることが必要だと考えられる．
@Homeではこのような状態に対処するために，ロボットの前後に障害物が存在しない場合には微速で前進もしくは更新するリカバリビヘイバを開発し使用している．
つくばチャレンジにおいても，このlead_outプラグイン(緊急脱出用リカバリビヘイバ）を導入するべき．

iou16, 2017-02-14:
@Homeでは後方視界用のURGがあったが, 今回はないのでどうするべきだろうか.
『コストマップを見れば良いのでは？』と思うかも知れないが, lead_outの当初の目的は“反射動作を実装”するというものなので, できる限り“センサ情報→反射動作”的な実装をしたい.

Nasupl, 2017-02-14:
今回の場合は前進するだけでも解決した可能性があるし、前進だけでもできたほうが良いのでは。
スキャンがある方向だけに動けないもんですかね。

iou16, 2017-02-22:
現在開発中のrecovery behaviorの挙動(4倍速).
[image]
具体的には, わざと自己位置を壁上にずらすことでpath plannerをabortさせ, 開発したrecovery behavior(前方を2D LIDARで確認し開けていたら直進するだけ)で復帰している様子.

closed 2017-03-07 by Nasupl