open-rdc/orne_navigation #240: 最低加速度の設定

Nasupl, 2016-11-16:
icart_mini_controlでは，最低加速度を設定することが可能だが，現在は設定されていないため速度指令がでなくなった場合急停止する．このパラメタを設定することでこの問題を解決する．

Nasupl, 2017-02-23:
ORNE-αは勝手にゆっくり止まるので必要なさそう．βの狭幅化が終わったらβで調整する．

iou16, 2017-04-27:
https://github.com/open-rdc/icart/blob/indigo-devel/icart_mini_control/config/icart_mini_control.yaml#L34
↑このファイルをこのまま変更するか, 必要であればαとβで分けるか, 意見お願いします.

Nasupl, 2017-04-27:
@iou16
そもそもβにおいてはこのパラメータ有効に機能するの？

iou16, 2017-04-27:
βには効き目ありそうですよ

Nasupl, 2017-04-27:
@iou16 
βで有効なパラメータをαで使ったときに、αの動作に影響がでないならそのまま変更すればよいのでは

iou16, 2017-04-27:
それもそうだな、影響なさそうなのでこのまま変更する.

iou16, 2017-05-17:
最低加速度の設定前の挙動
[image]

iou16, 2017-05-17:
最低加速度の設定後の挙動
[image]

yasuohayashibara, 2017-05-23:
あまり加速度を下げすぎると応答が遅くなり障害物回避などに問題が発生するかもしれませんが，オドメトリの精度を考えると最大加速度はできるだけ下げたほうが良いです．

Nasupl, 2017-09-18:
マージした (https://github.com/open-rdc/icart/pull/117)

closed 2017-09-19 by buzzshima