open-rdc/orne_navigation #250: 新マシン選定

Nasupl, 2017-04-11:


Hiroki-Goto, 2017-04-18:
早めに候補をあげる（備品扱いにする）
CPU：i7-７世代以降
RAM：最低8GB，できれば16GB
SSD　容量は常識的なおおきさ
おおきさ：βのマシン以下（重さはとわない）
GPUはできればほしい
USBポート：3ポート以上
画面：取れたり180°以上回したい

buzzshima, 2017-04-18:
http://www3.lenovo.com/jp/ja/notebooks/thinkpad/p-series/ThinkPad-P40-Yoga/p/22TP2WPWP40

Nasupl, 2017-04-27:
Quadro M500MもちゃんとLinux向けdriverがあるっぽいのでP40 Yogaでいい気がしてきた

Nasupl, 2017-05-18:
見積もり中

yasuohayashibara, 2017-05-25:
購入の手続きを進めていますが，申し訳ありませんが，既にウェブページがなくなっています．
発注にはカタログが必要になるのですが，もしかして販売終了でしょうか．
もし分かる方はできるだけ早めに教えてくれると助かります．
このままだと再度型番選定から始めなければならないかもしれません．

Nasupl, 2017-05-25:
@yasuohayashibara 
Webページがなくなっているというのがよくわからないのですが、どのWebページがなくなっているのですか？
製品のWebページは存在しているようですが。 http://www3.lenovo.com/jp/ja/notebooks/thinkpad/p-series/ThinkPad-P40-Yoga/p/22TP2WPWP40

yasuohayashibara, 2017-05-25:
￥182,218という価格が記載されていますが，間違いないでしょうか．
見積もりが224,070円ですが，差額は手数料？
カタログと見積もりを事務に出して妥当性を審査されますので弾かれるおそれがあります．

Nasupl, 2017-05-25:
@yasuohayashibara 
今回見積もりをお願いしたのは http://www3.lenovo.com/jp/ja/static/catalog/nb-2016-p40yoga_cf_0119 の20GR0003JPで、このページには見積もりを取った製品の直販価格も書いてありますが、このページでよろしいですか？

yasuohayashibara, 2017-05-25:
見れました．ありがとう．
こちらで発注します．

yasuohayashibara, 2017-05-29:
転送します．

2017年5月29日 17:48 池田　大輔 <d-ikeda@chibasokki.co.jp>:
> 千葉工業大学
> 林原先生
>
> いつもお世話になっております。
> 本日用度課様よりPC発注の連絡を頂きました。
> ありがとうございます。
> 納期は確認でき次第ご連絡致します。
> よろしくお願い致します。
>
> ************************************
> 株式会社 千葉測器 船橋営業所
> 池田 大輔

closed 2017-05-30 by Nasupl