open-rdc/orne_navigation #261: ORNEの環境構築の更新

Hiroki-Goto, 2017-07-07:
ROSのバージョンがkineticの場合，今までの導入方法だとパッケージ等が足らないためcatkin buildできない問題が発生する

そのためwikiを変更する必要がある

Hiroki-Goto, 2017-07-07:
忘れないうちにメモ
・hokuyo_nodeの入れ方
http://answers.ros.org/question/243232/how-to-install-hokuyo_nodeurg_node-on-kinetic/
を参考に導入する
・icart_mini_driverのCMakeList.txtの24行目YPSpur->ypspur
・icart_mini_gazeboのCMakeList.txtに「add_definitions(-std=c++11)」の追加
・navigationパッケージのvoxcel_gridディレクトリの削除（？）
・YPSpurの導入時にcatkin build を行わない（依存関係の解決時に問題が起きる）

Nasupl, 2017-07-07:
@Hiroki-Goto navigation stackはkinetic-develなら通るのではないですかね？（膨張リセットは無効になりますが・・・。）

Hiroki-Goto, 2017-07-11:
ｔｆ周りの対応

ShotaroTerato, 2017-08-16:
旋回時orne-xのgazeboのモデルの動きがおかしかった（直進走行は正常）
→base_linkから駆動輪が前後方向に大きく離れていると信地旋回ができなくなる挙動がみられた
orne_x.xacro：ロボット本体､車輪ごと0.1m前方に移動させ、駆動輪軸をbase_linkに近づけたところ改善した

ShotaroTerato, 2017-08-16:
bringupのみのときのtf
nav_static_mapしたときのtf
[frames_only_bringup.pdf](https://github.com/open-rdc/orne_navigation/files/1227564/frames_only_bringup.pdf)
[frames_with_map.pdf](https://github.com/open-rdc/orne_navigation/files/1227565/frames_with_map.pdf)

ShotaroTerato, 2017-08-17:
https://answers.ros.org/question/259341/frames-not-shown-using-view_frame-node-but-in-rviz/
このページを参考に
rosrun tf2_tools view_frame.py
を実行した結果
[image]
配信されているtfトピック
/tf
/tf_static

Nasupl, 2017-08-22:
@Hiroki-Goto 
https://github.com/open-rdc/orne_navigation/issues/261#issuecomment-313697900 について，kinetic-develブランチを作成して反映してください．

Nasupl, 2017-08-24:
具体的作業
- [ ]  orne_pkgs.installの変更
- [ ]  navigationパッケージの対応
  * https://github.com/open-rdc/navigation/issues/7
- [ ]  icartパッケージの対応
  * icart_mini_driverのCMakeList.txtの24行目YPSpur->ypspur
  * icart_mini_gazeboのCMakeList.txtに「add_definitions(-std=c++11)」の追加
- [ ]  #269

Hiroki-Goto, 2017-09-05:
rvizの読み込むファイルを元に戻す

Nasupl, 2017-09-21:
@Hiroki-Goto icartパッケージにもkinetic-develを立ち上げたので，CMakeLists周りの変更をお願いできますか？

Hiroki-Goto, 2017-09-21:
@Nasupl プルリクを送りましたので確認をお願いします.

Nasupl, 2017-09-21:
@Hiroki-Goto 動作はKinetic環境で確認済みですよね？

Nasupl, 2017-09-26:
https://github.com/open-rdc/icart/pull/119 と https://github.com/open-rdc/icart/pull/118 でicart周りは終わり

closed 2017-10-17 by buzzshima

Pethon, 2018-08-23:
ROS kinetic用のorne環境をインストールスクリプト
[https://github.com/Pethon/setup_new_machine/tree/master/bash](url)