open-rdc/orne_navigation #266: ORNE-Xの外装

Hiroki-Goto, 2017-08-08:
林原先生の夏休みの宿題

yasuohayashibara, 2017-08-30:
本日，工作センタと相談して，最終的に以下の方法で製作することにしました．

１）CADで外装の3Dモデルを製作する．
２）輪切りにして，2Dのデータに変換
３）2Dデータに従いスタイロフォームをウォータジェットで切断
４）重ねて段差をヤスリがけ　→　型が完成
５）その型を使用してFRP成形
６）乾燥したら，スタイロフォームを崩して完成

FRP成形をしたことがないので，本日キットを購入して小さなモデルで勉強会の後に試してみたいと思います．
ご興味のある方はご参加下さい．

yasuohayashibara, 2017-08-31:
お試しでサーフボード補修用のFRPを購入した．

使用したキットの写真
[image]

スタイロフォームに凹みを作り，グラスウールをかぶせて，硬化剤を入れたプラスチックの溶液を叩くように染み込ませた．
（作業を行っていたので写真なし）

アセトンでのハケの洗浄が遅れた(30分程度)ため，固着して再利用不可能となった．
紙コップの液体も30分程度で表面は完全に固形化した．
FRPの溶液は臭いが，どちらかというと洗浄に用いたアセトンの方が臭いがきついかもしれない．

１時間程度放置したら，臭いがそれほどしなくなったため，ゴミ袋にいれて，３Fに置いた．
２４時間放置して，完成の状態を確認します．

yasuohayashibara, 2017-08-31:
ベタな設計ですが，外装の外観です．
内部の部材が見えていますが，あまり気にしないでください．
すっぽり被せるように取り付けます．
干渉する部分は成型した後にカットします．

[image]

[image]

来週には製作する予定ですので，何かありましたら早めにご意見下さい．

Nasupl, 2017-08-31:
@yasuohayashibara 
前輪が回転した時に干渉しませんか？
また，スイッチパネルはどこにつく予定ですか？

yasuohayashibara, 2017-08-31:
前輪に関して
見た感じギリギリのような気がしますが，前輪は前後左右に動かすことができるので，干渉したら少しずらします．

スイッチパネル
手押し棒の根本の下に置きます．パネルの部分に関しては，干渉しますのでカットします．

工場に加工をお願いしなければならない関係で，先に外装の形状を確認させてもらえればと思います．

スタイロフォーム発注済み，工場にカットの予約もしました．
40mmのスタイロフォームが望ましいということですので，14枚のスタイロフォームを重ねることになります．

Nasupl, 2017-08-31:
@yasuohayashibara 
これは個人的な興味なのですが，輪切りはどの軸に対して輪切りにするのですか？

yasuohayashibara, 2017-09-01:
左右方向に積層します．
全て同じ形状でカットしてもらい，ロボットの左右の丸みはヤスリがけで作る予定です．

yasuohayashibara, 2017-09-01:
スイッチパネルを含めた外観です．
[image]

インターフェイス及びスイッチは以下の通りです．

1)有線LAN x 1
IE-FCM-RJ45-FJ-P

2)USB x 3
IE-FCM-UAB-A

3)LED
DW-10ML-IP-AC/DC5V-60V-R/C

4)スイッチ
M-2012W/B

5)電源
D/MS3100A12S-3P

スイッチとLEDに関しては，Arduinoを介して利用することになると思います．
全て防水ですので，常に暴露した状態で使用します．

yasuohayashibara, 2017-09-01:
整形したFRPに穴があり，水が漏れる様子
https://www.youtube.com/watch?v=A4W5yGAT3og&feature=youtu.be

yasuohayashibara, 2017-09-19:
スタイロフォームが到着したので，工場に出してウォータで型を作ります．

yasuohayashibara, 2017-09-20:
工場に発注しました．
[outer_body.zip](https://github.com/open-rdc/orne_navigation/files/1318323/outer_body.zip)

できるだけ明日(9/21)加工するそうです．

yasuohayashibara, 2017-09-30:
FPR成形しました．
ちなみに初めてFRP成形したので，ややミスありです．

１）ウォータカッターで切った型を抜きます．

[image]

２）重ねます．
　　CADデータに従って切れるので，もっと自由な形状にできるのですが，今回はわりとシンプルな形状です．　

[image]

３）ガムテープでぐるぐる巻きにします．
　　ちなみに，ガムテープを付けないとスタイロフォームが溶けていたので，隙間のないようにすることが必要だと思います．

[image]

４）両面テープでガラスマットを取り付け

[image]

５）樹脂をローラで塗って完了
　　ガラスマットがしなしなになって表面に張り付いていきます．
　　隙間にスタイロフォームを無理やり入れていますが，これは樹脂を含んだガラスマットが重力に負けて落ちてきたため，窮余の策として入れました．
　　おそらく２回に分けて向きを変えながら塗らなければならないケースだったようです．

[image]

今回樹脂を2kg使用しましたが，厚さを確認して，薄いようなら樹脂を重ね塗りしなければならないかもしれません．

buzzshima, 2017-10-03:
白で塗装

yasuohayashibara, 2017-10-04:
１）乾燥した様子

[image]

２）スタイロフォームを外した様子
　　準備と後片付け合わせて３時間かかりました．

[image]

３）硬さを確認した様子

https://youtu.be/G0-p3yR42zc

思っていた以上に固くてそのまま使えそうです．
まだ内側がしっとりしているので，もう少し乾燥させてから切断・塗装します．

得られた知見
・４日間経っても中はまだ乾燥していない．（においもきつい）
・服と靴が汚れる．靴と床がベトベトになる．保護メガネと手袋も必須．
・ノミのような工具があると良い．
・スタイロフォームが溶けていた．
・ガムテープは正解．スタイロフォームの溶け防止と離型の面で役に立った．

yasuohayashibara, 2017-10-10:
グラインダで不要部分を削り取りました．
[image]

クレンザでケースを洗いましたが一部繊維が飛び出しており，まだチクチクします．

パテを盛って，ヤスリをかけて，必要な部分に穴をあけたら塗装します．

Nasupl, 2017-10-10:
今週末に向けて
* まずは外装を機体に合わせる
* 外装が使える場合
  * パテ盛ってヤスリがけする
  * 塗装する
  * PCは外装の下に入ってしまうので，無線LANルータを積んでvino, reminaでVNCできることを確認する
* 外装が使えない場合
  * 電子回路，PC周辺の防水処理をする

yasuohayashibara, 2017-10-11:
外装をロボットに被せてみた．
外装のバリがフレームに干渉してはめづらいが，フィットしている．
干渉する部分があったが，フレームをずらすと干渉しなくなった．
（長島君からいただいた図面と少しずれていたので，図面に従ってずらしました．）

[image]

yasuohayashibara, 2017-10-11:
１）ペンでマーキングして，現物合わせでカットして，再度被せた様子．
　　良い感じではまっています．（パテ塗りの作業の紹介が抜けていた．．．）

[image]

上から見た様子
[image]

２）ヤスリがけをしました．
　　（こちらも写真なし．２時間？ぐらいかけて穴あけヤスリがけをしました．）
　　粉を被っていたのでクレンザで洗いました．

[image]

３）４号館地下の塗装ブースに移動

[image]

４）塗装後の様子（表面に凹凸があるので見た目がいまいち）

[image]

５）乾燥室に放置（塗装ブースの隣に乾燥室があります）

[image]

あとは，実ロボットにはめて実際に使用できるかをチェックします．

yasuohayashibara, 2017-10-13:
ロボットにはめてみました．

注意事項として，

１）ほとんど乾いていますが，だまになっていて乾ききっていないところがあります．
　　よく注意して作業して下さい．
２）はめると中の部品にはアクセスできません．
　　再度あけて作業するのは結構大変ですので，よくチェックして組み立てる必要があります．
３）ポールと斜め付けURGはケースを取り付けてから組み立てる必要があります．
　　ケーブルの扱いが難しいですが，予め配線してから組み立てることになります．
　　※斜め付へURGに関しては，夏迫くんが外さなくても良さそうな処置をしていました．
４）あまり無理な力を加えるとおそらく壊れます．
５）バッテリの電圧メータが見えなくなります．

あと，根本的な問題として，組み立てて気づいたのですが，下からの水には弱いです．
想定されることとして，

１）雨粒が大きい時に，地面に叩きつけられた水が跳ね返って中の電子部品を濡らす．
２）車輪が雨水を跳ね上げて中の電子部品を濡らす．

どれも相当な雨にならないと起き得ないことだと思いますが，頭の片隅に置く必要があります．
これを防ぐためには，やはり電子部品及びPCは何かで囲う必要があると思います．

あと，外装ですが中に水が入ってしまった場合，内側の壁を伝って水が電子回路やPC側に流れ込むおそれがあります．
水が入る位置より下にあるものに関しては，基本濡れるかもしれないという気持ちでいたほうが良いです．

外観

[image]

穴がありますが，雨が降ったらこちらは布テープなどで閉じて下さい．

[image]

あと，USBの配線（モータドライバ及びIMU）が長かった(2m)であったため，0.5mに入れ替えました．
本日夏迫くんがチェックするということですが，念のためお伝えしておきます．

yasuohayashibara, 2017-10-13:
実験走行(10/14,15)後に本格的な防水処置及び端子台の追加をして，issueを完了させたいと思います．

Nasupl, 2017-10-17:
### 運用してわかったこと
* 取り外しにくい
* 端子台がないと困る
* GPSはβに乗せる

yasuohayashibara, 2017-10-24:
制御板を作成しました．

写真は製作途中のものです．
[image]

yasuohayashibara, 2017-10-26:
本体に取り付けました．

[image]

これに合わせて外装をカットします．

yasuohayashibara, 2017-10-27:
外装をカットして，ポリカーボネートの窓を付けました．

[image]

一気にそれらしくなりました．
あとはコーキングして終了です．

Nasupl, 2017-10-27:
@yasuohayashibara 制御盤と外装の間の隙間になにか処理する予定はありますか？

yasuohayashibara, 2017-10-28:
ありません．
とりあえずは養生テープかなと思います．
あと，制御盤と外装の間は浸水してもそれほど大事に至らない気がします．

Nasupl, 2017-10-28:
@yasuohayashibara とりあえずまだ利用できる状態ではないのでそれで問題ないと思いますが，制御盤裏面側に出ている端子の短絡が不安です．

yasuohayashibara, 2017-10-28:
裏面端子に関しては，配線し終わったらコーキングしようかと考えています．

yasuohayashibara, 2017-10-28:
FRPとポリカーボネートの間をコーキングしました．
明朝までには乾いていると思いますが，扱いには気をつけて下さい．
（本体には注意書きを付けました．）

[image]

yasuohayashibara, 2017-10-28:
制御板背面のコーキングは残っていますが，外装に関しては一通りの作業が終了したので，issueを閉じます．

closed 2017-10-28 by yasuohayashibara

yasuohayashibara, 2017-10-31:
電源のコネクタが届きました．

[image]

yasuohayashibara, 2017-11-27:
以下に外装も含めたCAD図を置きました．
http://www.hayashibara-lab.it-chiba.ac.jp/member/project/%e3%81%a4%e3%81%8f%e3%81%b0%e3%83%81%e3%83%a3%e3%83%ac%e3%83%b3%e3%82%b82017/orne_x/170831orne_x_%E5%A4%96%E8%A3%85.zip