open-rdc/orne_navigation #272: StateTriggerPanelをkineticで読み込むとrvizが落ちる

Hiroki-Goto, 2017-09-19:
kineticでnav.rvizを読み込んだ場合，おそらくStateTriggerPanelが原因でrvizが落ちる

Hiroki-Goto, 2017-09-19:
今のところの対応
kineticはnav.rviz以外のファイルを読み込むことで対応
自律移動する際はサービスを送ってる（wikiのコマンドではスタートしない）

Nasupl, 2017-09-19:
@Hiroki-Goto orne_toolsを最新にしてビルドしましたか？

Hiroki-Goto, 2017-09-19:
最新にしてみてやってみた結果です
・orne_toolsのSaveTriggerPanel->読み込むことはできた
・orne_rviz_pulginsのStateTriggerPanel->rvizは落ちる

Nasupl, 2017-09-19:
@Hiroki-Goto Hiroki-Goto orne_rviz_pluginも#270 で対応しているはずですが、そのテストはどのブランチで試していますか？

Hiroki-Goto, 2017-09-19:
テストのブランチですが，change_orne_x_descriptionでやっています

Nasupl, 2017-09-19:
@Hiroki-Goto そのブランチには #270 がマージされていないからだと思います。なぜKinetic develではなくそのブランチなんですか？

Hiroki-Goto, 2017-09-19:
kinetic develのブランチはαのURDFの変更などが適用されていなかったため，変更を適用しているchange orne x description でテストをしていました

最新版のindigo develでやってみしたが落ちませんでした

Nasupl, 2017-09-19:
@Hiroki-Goto indigo-develをkinetic-develにマージしてください

Nasupl, 2017-09-20:
#273 が終了したら終了

Nasupl, 2017-09-20:
#273 マージした

closed 2017-09-20 by Nasupl