open-rdc/orne_navigation #292: 自動ドア対策

Nasupl, 2017-10-17:
自動ドアの手前で止まって経路を他のところに生成しようとしてしまう
戦略面で対応．自動ドアエリアではドアが開いたら強制的に進むようなプログラムを書く．

iou16, 2017-10-20:
fulanghua_waypoints_nav側を改造&デバッグ中.
waypointsをserviceで受け渡し出来るようにしている. 
https://github.com/open-rdc/fulanghua_navigation/tree/add_wp_serv

Nasupl, 2017-10-20:
@iou16 fulanghua側の改修、 https://github.com/open-rdc/fulanghua_navigation/commit/eb2bf46e0cd2765a8d985d3d4ef94beb0324700f で終わりですか？

iou16, 2017-10-20:
@Nasupl 変更すべき箇所はこれで終わりかと. でも, 動作確認というかデバッグが終わっていません.

Nasupl, 2017-10-20:
@iou16 了解。明日試します。

Nasupl, 2017-10-24:
https://github.com/open-rdc/orne_navigation/tree/add_%23292 で対応中

iou16, 2017-10-25:
取り敢えずそれっぽいものができたので, 明日, 2号館1階の自動ドアで試す.
@homeで使っていたただまっすぐ進むだけのserviceは現状使用していない(自動ドアの目の前で止まって, 開いたらwaypoints_navを再開するだけ)だが, 明日の結果次第では導入するかも.

iou16, 2017-10-26:
2号館1階の自動ドアで試したが, 一応通れた. その際の動画は,  @buzzshima がアップロードしてくれる(らしい).

buzzshima, 2017-10-26:
https://youtu.be/R91wTuLpTMU

これがURLです

buzzshima, 2017-10-31:
コストマップを編集したらうまくいったのでとりあえず保留

iou16, 2017-11-01:
https://github.com/open-rdc/orne_navigation/blob/add_%23292/orne_strategy/scripts/tsukuba_challenge#L116-L117
使用するかしないかは別として, この部分の実装は修正しないといけない
具体的には
- LaserScanをPointCloudに変換
- global座標系へ変換
- 自動ドアの周辺の一定範囲にPointCloudが含まれていないかを確認(実装方法要検討)

closed 2017-11-07 by Hiroki-Goto