open-rdc/orne_navigation #315: 空気圧の確認(ついでにオドメトリ)

buzzshima, 2017-10-31:
ホイールオドメトリの精度が空気圧にどれくらい依存しているかの確認

buzzshima, 2017-11-01:
両機体ともに適正空気圧(210ｋpa)より下がっていて適正まで空気圧を入れた．(そのときbetaはもともとの空気圧が低すぎたためか空気を入れた時機体がジャッキアップしたみたいに上がった) 
また空気圧の調整前後でのホイールオドメトリのチェックした．

実験は簡易的行った．
- 2号館3階を一周し計測
- 調整前後で同じ速度で走行したが経路は異なる(コースと始点と終点は同じ)

調整前は赤いライン
調整後は黄緑ライン
alphaのホイールオドメトリ
[image]
＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿
betaのホイールオドメトリ
[image]

# 感想
両機体ともに空気圧を適正のほうが真値近くなったがだいぶかけ離れている．alphaはパラメータの設定ミスを疑うほどずれている．

Nasupl, 2017-11-01:
@buzzshima 
現在のαのパラメータ，
RADIUS[1] 0.148569 # 0.148493 #0.150044 #0.149044 #0.150
TREAD 0.56 #old_0.449592 #0.450
ですが，合ってますでしょうか？

yasuohayashibara, 2017-11-02:
#319 でコメントしてしまいましたが，こっちに空気圧の議論がありました．
同じ圧力でこれだけずれるのですね．
意図的に圧力を変えてずらすのもあまり良くないと思いますので，先ほどの空気圧の議論は無しで．

左右のホイールの半径の設定があるのであれば，それで対処できそうですね．

closed 2017-11-07 by Nasupl