open-rdc/orne_navigation #319: alphaの機体パラメータの変更

buzzshima, 2017-11-01:
#315 でαのホイールオドメトリがオドメトリがあまりにもずれていたので機体のパラメータを見たら違っていたのでパラメータを変更して計測した．
[image]
＿＿＿＿＿＿＿＿＿＿
変更前のやつ
[image]

変わってないような............

Nasupl, 2017-11-01:
マシにはなっているのでは

yasuohayashibara, 2017-11-02:
どちらにしてもひどいですね．100m走行すると90度以上ずれる計算ですね．
（修正前のパラメータだと180度以上）

つくばチャレンジの2kmのコースを走行すると5回転(1800度)ほどずれる計算です．
まあ実際にはIMUで方位を求めているので問題ないかもしれませんが，できるだけ直進できるようにホイールの空気圧を調整したほうがよいと思います．
（もしくは左右のアンバランスを調整するパラメータがあれば，その修正でも構わないかもしれませんが．．．）

このような基礎的なデータは問題がどこにあるかを明確にできるので重要ですね．

Hiroki-Goto, 2017-11-21:
@buzzshima  がタイヤの径のパラメータを変更して実験する