open-rdc/orne_navigation #335: 地図の解像度を粗くしてみる

iou16, 2017-11-06:
現在の占有格子地図の解像度は0.05[m]だが, 屋外で自律走行するならそこまで細かくなくていいのかも知れない.  少し試してみたい. どの程度の解像度が適切だが要検討.

buzzshima, 2017-11-14:
画像編集ツールで試してみる

taishiyamamoto, 2019-11-12:
植田君にも担当してもらうことになりました。
memberに招待した後、assignする。

hoshianaaa, 2019-12-06:
gmappingのパラメータを変更すれば解像度を変更できます
[image]
orne_navigation_executor/build_map_teleop.launch内

taishiyamamoto, 2020-02-21:
地図のリサイズを1ピクセル辺り0.1mの状態で自己位置推定を行わせて見ました．
わかったこととして，amclで占有格子地図を元に尤度場を生成する際の時間が短くなりました．
自己位置推定の精度に関してもパーティクルの広がりは従来（0.05m/pixel）のときとあまり大差ないように思いました．
個人的には地図の解像度は0.1mの分解能で生成しても特に問題ないと考えております．
https://youtu.be/VPx8Lzc-A-U

yasuohayashibara, 2020-02-22:
分解能0.1mの採用賛成します．

tiger0421, 2020-02-25:
今後のマップの解像度は0.1mと決定しましたので，クローズします．

closed 2020-02-25 by tiger0421