open-rdc/orne_navigation #337: 後方視界の検討

Nasupl, 2017-11-07:
障害物の回避や衝突時の復帰にはやはり後方視界が必要だと考えられる．以下のようなセンサを搭載し後方視界を確保することを検討する．
以下に考えられる方法を例示する．

* 安価な2D LIDAR
[RP LIDAR](http://www.robotshop.com/jp/ja/rplidar-a2-360-laser-scanner.html)のような安価で小型なLIDARを搭載する．
ロボットの後方に障害物があるのかないのかさえ分かれば十分なので，安価な製品で十分．安価な製品といえど$500近くするのが難点．

* 1Dのレーザー距離センサ
[VL53L0X](https://www.switch-science.com/catalog/2894/)のような距離センサを複数搭載する．ROSで使用するための開発コストが大きい．視野が狭いので複数取り付けが必要．

* 超音波センサ
[MaxSonar](http://akizukidenshi.com/catalog/g/gM-08238/)のような超音波センサを搭載する．レーザー距離センサと比較すると指向性が低いので，１個のセンサで広い領域をカバーできる（どの方角にあるかはわからないが）のがメリット．これもROSで使用するための開発コストがは大きい．

yasuohayashibara, 2017-11-09:
ミーティングで話しましたが，もう一つTOP-URGを搭載するという方法があります．
ロングレンジの測域センサを付けることで，自己位置推定の安定性も上がります．

Nasupl, 2017-11-21:
Top-URGとRPLIDAR A2を比較できるようなデータを取得した
https://photos.app.goo.gl/1R4CzvfMxS0llCmk2
RPLIDARのレンジは6m程度であった．また，Top-URGと比較してデータのばらつきが大きいことがわかる．

buzzshima, 2018-07-20:
@Pethon 
取り付け用パーツ完成
取り付け位置 姿勢(前方urgとくらて)
xとyは同じでzが- 400mm
yaw軸に180度回転

でこの位置でurdfを書いて下さい

斜めURGを参考に

buzzshima, 2018-07-20:
後ろ向きURGのものを加工した(取り付けジグを追加するかも)
位置と姿勢は baselink

x -340  y 0 z 0.22
yとｚは前方urgと同様
取り付け位置

urdfは斜めurgの参考(/orne_description/urdf/orne_alpha.xacro)   変更はβのみorne_description/urdf/orne_beta.xacro) 行う
#249 が似たもの

yasuohayashibara, 2018-07-20:
できれば図や写真を載せてください．

buzzshima, 2018-07-24:
センサだけ情報を追加したから妥当だと思います

取り付け板とかの情報書いてないですよね？

Hiroki-Goto, 2018-07-24:
あとicart_middle.gazeboも変更を行う

Hiroki-Goto, 2018-09-20:
@Pethon が後入れの発注する

buzzshima, 2018-10-04:
取り付け完了
[image]

Pethon, 2018-10-11:
後ろのlidarを利用して自己位置推定できるのをシミュレーター上で確認しました．
[https://youtu.be/1wfOm5yBxzc](https://youtu.be/1wfOm5yBxzc)

Pethon, 2018-10-11:
実機を用いて、実験を行いました。
動画[https://youtu.be/wluyvotoV8k](https://youtu.be/wluyvotoV8k)

yasuohayashibara, 2018-10-13:
人が後ろからついていくと，後ろにコストが残りますね．
つくばチャレンジでは問題にならないと思いますが，屋内の自律移動のときには問題になるかもしれません．
コストを残す時間を短くしたほうが良いかもしれません．

buzzshima, 2018-10-25:
＠buzzshima レビューして閉じる

closed 2018-10-25 by buzzshima