open-rdc/orne_navigation #341: 電装の改良

Nasupl, 2017-11-07:
電装の全体的な改良が必要
* 電源周辺
  * 12V電源の生成方法
  * 搭載したままでのバッテリー充電
* 緊急停止周辺
  * リレーが電力を食い続ける
* 意思表示関連
  * 状態を報せたい
      * パトランプ，スピーカー等
* 制御盤
  * スイッチ関連の配線

Nasupl, 2018-03-06:
### 緊急停止周辺について
モータドライバの駆動用電源に低ESRキャパシタを入れれば誘導電流によってモータドライバが破壊されたり，手押ししている時にガタガタとすることはない．
遮断回路は更なる安全策として有用だが，回路が複雑化しメンテナンス性が低下するため廃止する．

Hiroki-Goto, 2018-04-17:
音声
・ソフトがいいのがあれば青木に相談する
・ @buzzsima の音声をつかう（第一候補）

Hiroki-Goto, 2018-04-23:
音声に関しては #349 と別のissueとしてたてた

Hiroki-Goto, 2018-04-24:
@daiki-shibahara 電圧計の消費電力を確認する

daiki-shibahara, 2018-04-24:
[image]
この画像の回路の製作を行う。
現時点では、Ltspiceを使ったシミュレーションを行っている

daiki-shibahara, 2018-05-15:
プリント基板での作成が完了したため,動作実験を行う.

daiki-shibahara, 2018-05-29:
切削基板での動作確認が完了したため,プリント基板の発注を行う.

Hiroki-Goto, 2018-07-10:
ellecrowで発注してから閉じる
届いたら動作テストする

daiki-shibahara, 2019-04-09:
発注しました

atsushi10mino, 2019-05-21:
足りない部品を購入する

daiki-shibahara, 2019-06-04:
部品の購入を行い，基板を取り付け動作確認が取れたため，閉じます．

closed 2019-06-04 by daiki-shibahara