open-rdc/orne_navigation #353: 経路計画（+navigation）

Norihiro02, 2018-05-10:
取り敢えずturtlebot_navigationでは、経路が塞がれても自動で再度経路計画をおこなうということが分かった。
↓
どの様に移植するか。

Norihiro02, 2018-05-14:
turtlebot_navigation
https://github.com/turtlebot/turtlebot_apps/tree/indigo/turtlebot_navigation

yasuohayashibara, 2018-05-14:
後で見返した時に，そのときの状況が分かるように，できれば動画をYoutubeなどにアップしてリンクを張っておいてもらえると助かります．
言葉だけだと忘れてしまいますので．．．

> 取り敢えずturtlebot_navigationでは、経路が塞がれても自動で再度経路計画をおこなうということが分かった。
> ↓
> どの様に移植するか。

Norihiro02, 2018-05-15:
turtlebot_navigationの動画です。　
https://www.youtube.com/watch?v=LPhcNcJsR5w&feature=youtu.be

Hiroki-Goto, 2018-05-15:
https://github.com/open-rdc/fulanghua_navigation/blob/indigo-devel/fulanghua_waypoints_nav/src/waypoints_nav.cpp
にclear costmapのサービスがあるっぽい

Norihiro02, 2018-05-19:
turtlebotとorneの比較してわかった事
1. 使用しているplanner系のplaginは同じ
2.  waypointの間隔が異なる
　turtlebot：実験時に設定しているのはstartとgoolだけ
　orne:細かくwaypointを設定している(数メータ単位)

現状ではwaypointを細かく設定し、1つの経路としている。
　別のwaypointを設定しないと別経路を生成することができない。

Norihiro02, 2018-06-14:
公園内に関しては
チェックポイントをnode
道をedge
とするグラフを作成し、各nodeに位置情報を加える。
このグラフを使用すれば「任意のチェックポイントを通る最短経路」を生成することができる。
さらに各node(チェックポイント)の位置情報をwaypoinファイルに出力すれば現状のシステムでも課題達成できそう。

Norihiro02, 2018-07-10:
5分前ルールは無し（こっちの思い込みであった）
現状：ロボットは一つの経路をたどる事しかできない。
やりたい事：ロボットが任意の経路を選択できるようにする。

以前作ったwaypointファイル生成プログラムを移植して作る

Norihiro02, 2018-08-07:
グラフは手動で作り、以下のようにnavigationをすることができた。
https://youtu.be/5KtCJokeFvY

Hiroki-Goto, 2018-08-07:
計算に関しては一旦閉じる

closed 2018-08-07 by Hiroki-Goto