open-rdc/orne_navigation #354: gammaの制御法の方針に関して

yasuohayashibara, 2018-05-15:
gammaに関しては機械学習を用いて自律走行させる予定であるが，そのアイディアを考える．

yasuohayashibara, 2018-05-15:
まずは論文を読む．
・PoseNet: A Convolutional Network for Real-Time 6-DOF Camera Relocalization
・GOSELO: Goal-Directed Obstacle and Self-Location Map for Robot Navigation Using Reactive Neural Networks
・End to End Learning for Self-Driving Cars

yasuohayashibara, 2018-05-29:
従来の方式を用いるということでは以下が考えられる．

■方式１　End-to-End
NVIDIA BB-8の方式
カメラとステアリングのデータを事前に取得
→ 深層学習で　（入力）カメラ画像→（出力）ステアリング　を学習
障害物の検出・停止に関しては，URGを用いる
問題：当日は環境が大きく変わる
もしかしたら，全天球カメラを用いることで地上の外乱に影響されないような経路が生成できるかも？

■方式２　自己位置推定
posenetの方式
カメラ画像と位置・方位データを事前に取得
→ 深層学習で　（入力）カメラ画像→（出力）位置・方位　を学習
自己位置推定以外の部分に関しては，従来のmove_baseを利用

■方式３　その他
画像だけではなく，URGやIMUの値も含めて学習

JinyaIseki, 2018-05-30:
方式3に関連する研究
https://arxiv.org/pdf/1803.10158.pdf
入力　画像、GPS,map,
出力　steering angle,Speed
ニューラルネットワークの層の種類
CNN,LSTM,FN

URG,IMUを使ったものはなさそう

JinyaIseki, 2018-07-10:
https://arxiv.org/pdf/1803.10158.pdf
の方法を参考に自律走行するhttps://arxiv.org/pdf/1705.08639.pdf

公園内ではカーナビゲーションを利用できない
→GPSの履歴を使う　もしくは現在地、目的地の記載してある地図を入力し、LSTMに大域的経路計画をさせる
　→LSTMは　https://arxiv.org/pdf/1807.01281.pdf　https://arxiv.org/pdf/1705.08639.pdf　を参考に設計する

様々なセンサーの値を入力する
→IMU、エンコーダー、レンジセンサなど

学習用データの経路から外れた時に自律走行できるのか？
→学習用データを増やす、カメラを３台用意する

yasuohayashibara, 2018-07-24:
本日のミーティングでも議論しましたが，自律走行に関しては，
入力：全天球カメラの画像
出力：目標速度ベクトル（ジョイスティックの操作）
として，まずはデータを採取する．

案は２つあり，
１）画像に対する教師データをオフラインで与える方法
２）PFoEを使用する方法
とする．

１）に関しては，オフラインで教師信号を生成して与える．
生成の方法としてVRを利用して道の中心を与えることにより，望ましい目標速度ベクトルを求めることを検討している．

yasuohayashibara, 2018-07-24:
議論が終了したのでissueを閉じます．

closed 2018-07-24 by yasuohayashibara