open-rdc/orne_navigation #356: 電装の改良（スイッチ版）

Hiroki-Goto, 2018-06-19:
逆起電力防止装置（#341 で作成したリレー）などがないとモータドライバの故障が頻繁に起きるため、防止用の回路（？）を作成する

スイッチでモータの線を切断する方向で作成

Hiroki-Goto, 2018-06-19:
必要な部品
- LED（径が大きいやつ）
- トグルスイッチ（４接点）

Hiroki-Goto, 2018-06-19:
トグルスイッチは最低10A耐電流あるもの

Hiroki-Goto, 2018-07-10:
#192 にあった回路を追加する

daiki-shibahara, 2018-07-19:
ミスミでは売っていなかったため,千石でセメント抵抗12個購入お願いします.
https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=2ADT-6UKM
リレーですが,γ用で使うかもしれないため,6つ購入お願いします.
https://jp.misumi-ec.com/vona2/detail/222000863520/?KWSearch=LY%20%E3%83%90%E3%82%A4%E3%83%91%E3%83%AF%E3%83%BC%E3%83%AA%E3%83%AC%E3%83%BC&searchFlow=results2products

daiki-shibahara, 2018-07-19:
[image]
#192 の追加回路作成し,セメント抵抗つけるだけです.
動作確認はその後行います.左:追加作成した回路   右:#192で作成した回路

yasuohayashibara, 2018-07-20:
ミスミで販売していますよ．
例）SQMW51R0J

リレーのリンク先ですが，１３５個の候補があるそうです．
[image]

daiki-shibahara, 2018-07-20:
リレーのリンク先間違えていました.以下を6個お願いします.
https://jp.misumi-ec.com/vona2/detail/222000863520/?HissuCode=LY2%20DC12
ミスミでセメント抵抗見つけました.5個単位での発注のため15個お願いします.
https://jp.misumi-ec.com/vona2/detail/222005596090/?KWSearch=%E6%8A%B5%E6%8A%97%E5%99%A8%20%E5%9B%BA%E5%AE%9A&searchFlow=results2products&relatedKeyword=1

yasuohayashibara, 2018-07-20:
リンク先を見直してください．
[image]

> ミスミでセメント抵抗見つけました.5個単位での発注のため15個お願いします.
> https://jp.misumi-ec.com/vona2/detail/222005596090/?HissuCode=SQBW2010RJ&PNSearch=SQBW2010RJ&KWSearch=SQBW2010RJ&searchFlow=results2products

daiki-shibahara, 2018-07-23:
リレー(LY2 DC12)を6個
セメント抵抗'(SQBW2010RJ)を15個ミスミで発注おねがいします.

yasuohayashibara, 2018-07-25:
シャーシマウントの抵抗になっていますが正解？
https://jp.rs-online.com/web/p/panel-mount-fixed-resistors/1614267/

yasuohayashibara, 2018-07-25:
こちら回答が来てから発注しますので，回答お願いします．
以下を見る限り，シャーシマウントの抵抗では取り付けにくいのではないかと思います．
https://github.com/open-rdc/orne_navigation/issues/356#issuecomment-406287103

daiki-shibahara, 2018-08-15:
[image]
回路の方作成し終わりました．
動作確認お願いします．

Hiroki-Goto, 2018-09-07:
αのほうには実装しました
[image]

daiki-shibahara, 2018-09-11:
電源を抜いているときは緩やかに押すように

closed 2018-09-11 by daiki-shibahara