open-rdc/orne_navigation #362: 地図の作成

Hiroki-Goto, 2018-07-10:
- [ ] 確認走行用
- [ ] 公園に行くまでの地図
- [ ] 公園内の地図
    - [ ]　固定経路用
    - [ ]　すべての経路用

Hiroki-Goto, 2018-07-10:
公園までの地図：7/１５まで
公園の地図：9/1

daiki-shibahara, 2018-07-19:
公園までの地図合成完了しました.

daiki-shibahara, 2018-07-19:
[kenkyugakuen_maybe_ok.zip](https://github.com/open-rdc/orne_navigation/files/2210956/kenkyugakuen_maybe_ok.zip)

daiki-shibahara, 2018-09-14:
[image]

daiki-shibahara, 2018-09-14:
[image]

daiki-shibahara, 2018-09-14:
公園内の地図合成は現状地図が足りないため，明日取得必要があります
市役所から公園内までの地図の合成はできました

daiki-shibahara, 2018-09-19:
9/15に市役所から公園までの地図で自律移動させた所
https://youtu.be/jHe5Dy7pzJs　
途中から地図の合成が上手くいっておらず新しく合成しなおした
[image]

yasuohayashibara, 2018-09-23:
新しい発見があったので，共有します．
先日1/10でrosbagを再生すると地図の精度があがるというtipsを教えてもらったのですが，一例ですが効果絶大です．

大学の芝生周辺の地図を作成した様子です．
https://youtu.be/4iisqSt_rYo

rosbagを1/10で再生して作成した地図です．
https://youtu.be/jkyltQa8-xo

一例ですが素晴らしく精度が上がりました．

yasuohayashibara, 2018-10-07:
どうしたら歪みの少ない地図が作成できるかを検討するために，以下のbagファイルで地図を作成した．
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2018-05-26_Experiment_Tsukuba_orne-beta/

確認走行区間のデータ（２周走行？）
2018-05-26-10-59-15.bag

※以下，別のPCで作業をして現在そのデータが閲覧できないため，後で追記します．
１）bagファイルからrosbag filterを使用して，/map, /map_metadata, /tfの/mapフレームを削除
　　すでに生成した地図のデータが入っていたが，再生成するためにはじめに削除

２）条件を変更しながら，gmappingで地図を生成
　　条件１　ジャイロオドメトリのみで作成
　　条件２　デフォルトのパラメータで作成
　　条件３　試行錯誤したパラメータで作成

結果
１）ジャイロオドメトリのみでもある程度地図が作成できる．
ただし，１周すると20deg程度のズレが発生するので，そのままでは使用できない．
逆にアナデバのIMUであれば，もっとズレの少ない地図が生成できるかもしれない．

２）gmappingのデフォルト値では歪みの無い地図は生成できない．
屋内を想定しているので，レーザレンジで検出される物体が少なすぎる．

３）ジャイロオドメトリに重きをおいたgmappingでは改善するものもあるが，一致はしない．
ということで，切り貼りの作業が発生していると思われる．
ただし，地図の切り貼りはある程度の整合性は取れるかもしれないが，全体的な歪をなくすことにはならない．

あと．実はTOP-URGは60mまで距離が検出できることを発見した．
※データには60m近い距離のデータが含まれています．
https://www.hokuyo-aut.co.jp/search/single.php?serial=21

解決策
１）/tfの/odomフレームを衛星地図を参照しながら手動で修正できる機能を実装する．
２）ジャイロのズレによるものであるため，よりドリフトしないジャイロに変更する．
　　アナデバのジャイロに変更したが，すでに解決しているかもしれない．

アナデバのジャイロを使用した以下のデータに関しても同様の作業を行い，地図のズレを検証する．
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2018-10-02/

daiki-shibahara, 2018-10-09:
確認走行区間から信号機までの地図合成したものです
https://drive.google.com/file/d/1tZYNv8spjAQ1Rbtjy99bBbCRV0-OwBWU/view?usp=sharing

yasuohayashibara, 2018-10-10:
以前から気になっていたIMUの精度が高ければ，ジャイロオドメトリだけで地図ができてしまうのではという疑問に関して調査しました．

結果は「できました．」

おそらくはレーザレンジセンサを用いたSLAMより幾何学的に正しい地図が描けているのではないかと考えます．
公園は十分な距離データが得られないことが原因と思いますが，gmappingで地図作成すると，かえって地図を歪ませてしまっていると思います．
（パラメータを調整すれば改善すると思いますが．．．）
なお，データが不足しているのかエラーによりgmappingが動きませんでした．

 以下地図生成のコマンド
（１行目は必ずしも必要ないが，とりあえずファイルサイズを小さくしたかったので．．．）

あと，rvizはrosbag playの前に立ち上げて下さい．hokuyo_linkのtfが最初に１回しか投げられないため，/scanがエラーとなります．（少々ハマりました．．．）

```
rosbag filter 2018-08-04-13-35-47.bag converted.bag "topic != '/velodyne_packets' and topic !='/velodyne_points'"
rviz
rosbag play converted.bag
```

[image]
黄色がSLAMを使用せずにジャイロオドメトリだけで作成した地図

元の画像ファイル
https://photos.app.goo.gl/ECQYbi5KBjB2Ect49

木や建物，道の形を見る限りほぼ完全に一致していると思います．
テーブルに見える丸い造形物だけずれていますが，もしかして移動した？
gmappingで地図を作成しているわけではありませんので，動的障害物（安全管理責任者など）も写り込んでいます．

yasuohayashibara, 2018-10-10:
以下のファイルに関してジャイロオドメトリを使用した地図を作成しました。
2018-10-02-14-24-20.bag （最近計測したbagfile）
http://www.hayashibara-lab.it-chiba.ac.jp/member/bags/2018-10-02/

作成して気づいたのは、

「オドメトリ」のパラメータに間違いがあるかも

です。
動画のホイールの様子を見ると、１０回転で１０m程度進んでいます。
確かホイールの直径は0.30m程度だったと思いますので、１回転0.62m程度進むと思います。
オドメトリの進みが実際より早いため、レーザの点が走行とともにぶれてしまいます。
なお、方向に関しては合っていそうな気がしますので、距離さえ調整できれば正確な地図が書けそうな気がします。

動画
https://youtu.be/zgRSvJx3ReA

作成した地図
[image]

ToshihiroNagao, 2018-10-25:
https://github.com/open-rdc/orne_navigation/issues/373#issuecomment-428826151
https://github.com/open-rdc/orne_navigation/issues/373#issuecomment-432906367
のコメントの通り地図の作成を行いました。

buzzshima, 2018-10-25:
作成済んだ

closed 2018-10-25 by buzzshima