open-rdc/orne_navigation #364: 人物探索

Hiroki-Goto, 2018-07-10:


buzzshima, 2018-08-21:
@ibukichi  進捗はどうですか？

ibukichi, 2018-09-17:
北陽3DLIDARが使えるかどうか試したところ接続が失敗したためかpointcloudで指定してもエラーが出て点群が表示されないです。 
[image]

ibukichi, 2018-09-19:
orne_bringupとhokuyo3dを立ち上げてもう一度実験したところ。
接続はできて、値も出てきている。
しかし、rvizに表示されない。
エラーを見てみるとtfの変換ができてないと出ていますが、tf＿treeを見た限り繋がっているのでtf内での変換がうまく言ってないorその他のところに飛ばしている可能性があることがわかりました。
センサ自体は最初赤点滅はするが、時間が経つと赤点滅は消える。つまり、センサは壊れてないかもしれないです。

ibukichi, 2018-09-19:
orne_bringupにあるhokuyo3d.launchをいじった結果、3次元点群が表示できました。どうやらvalueで指定したlink先がhokuyo3d_linkではなく、hokuyo_linkになっていたためでした。お騒がせして申し訳ございません。

ToshihiroNagao, 2018-10-11:
人物探索の実験動画です。
https://youtu.be/f7cLNp4Ccbg

ibukichi, 2018-12-06:
探索対象者へアプローチ動作の確認ができたため閉じる

closed 2018-12-06 by ibukichi