open-rdc/orne_navigation #369: 9/15まで必要品購入

buzzshima, 2018-08-21:
確認してコメント

buzzshima, 2018-08-21:
後入れナット

atsushi10mino, 2018-08-28:
SHNTP5-5
でいいですか？

atsushi10mino, 2018-08-28:
https://jp.misumi-ec.com/vona2/detail/110302607710/?HissuCode=PACK-SHNTP5-5

yasuohayashibara, 2018-08-30:
もし購入するのであればその旨伝えて下さい．

daiki-shibahara, 2018-09-05:
#356 よりリレー(LY2 DC12)を6個の発注お願いします
https://jp.misumi-ec.com/vona2/detail/222000863520/?HissuCode=LY2%20DC12
後入れナットは自分が把握している中では#337と#368で使う予定です

yasuohayashibara, 2018-09-07:
> #356 よりリレー(LY2 DC12)を6個の発注お願いします
> https://jp.misumi-ec.com/vona2/detail/222000863520/?HissuCode=LY2%20DC12

リレー発注しました．

yasuohayashibara, 2018-09-11:
リレー到着しています．

yasuohayashibara, 2018-09-20:
再度になりますが，明確な依頼が来てから発注になりますので，必要があれば型番と個数と価格をお願いします．

buzzshima, 2018-09-20:
後入れナットは @yasuohayashibara に頼み発注済

daiki-shibahara, 2018-09-29:
コーナーガード用のマジックテープが無い(#209 )と同じ購入(長さも同じくらいで)の仕方をする
@linevael が購入しに行く

daiki-shibahara, 2018-09-29:
αとβのタイヤ空気穴用キャップが無くなっています
α
[image]
先端部分の直径が7.5～8mm

β
[image]
5.5～6.0mm

ホームセンターなどで販売してそうなので @linevael に探してもらう

JinyaIseki, 2018-10-04:
γに非常停止スイッチを使用したので、予備の非常停止スイッチを２つ発注おねがいします
https://jp.misumi-ec.com/vona2/detail/222005759507/?KWSearch=AVN&searchFlow=results2products

Hiroki-Goto, 2018-10-11:
@linevael  どうなりましたか？？？？？

atsushi10mino, 2018-10-11:
https://www.monotaro.com/p/5136/9368/
おそらく後継品

linevael, 2018-10-12:
検収終わりました
[image]

yasuohayashibara, 2018-10-13:
> γに非常停止スイッチを使用したので、予備の非常停止スイッチを２つ発注おねがいします
> https://jp.misumi-ec.com/vona2/detail/222005759507/?KWSearch=AVN&searchFlow=results2products

予備は必要無いという判断で購入しない．

closed 2018-10-25 by buzzshima