open-rdc/orne_navigation #371: アナデバさんから提供されたIMUがうごかない

Hiroki-Goto, 2018-08-23:
アナデバさんから提供されたIMUの挙動調査だがwindows版では動いた

しかし、PCとIMUをつなぐ基板EVAL-adis2Zがlinuxに対応していないため、SPI-USBのものを別途購入する必要がある
使っていた基板はおそらく[これ](https://www.elefine.jp/SHOP/USB-ISS-SV.html)

yasuohayashibara, 2018-08-23:
Windows版って何？
具体的にソフトウェア名を記載して下さい．
また，後から見た人の参考になるように，できるだけ型番を示したほうが良いです．

yasuohayashibara, 2018-08-23:
以下のROSパッケージを用いて姿勢を検出する．
https://github.com/tork-a/adi_driver

Ubuntuで値を読み取るために以下を購入します．
Devantech USB-ISS-SV USB I2C・SPI・UART変換器

yasuohayashibara, 2018-08-23:
以下を１つ購入しました．
Devantech USB-ISS-SV USB I2C・SPI・UART変換器

動作確認の後，追加で２つ購入します．

Hiroki-Goto, 2018-09-07:
Devantech USB-ISS-SV USB I2C・SPI・UART変換器を使って動作確認を行いました

動作確認を行ったときのbagファイル
https://drive.google.com/open?id=1E5T_3gMpYTju9SjLjxeQOK3_5_LNheAA

yasuohayashibara, 2018-09-08:
動作の様子です．
https://youtu.be/uHPApUICayw

ADIS16470用に開発されたROSノードですので，そのままでは利用できないということかもしれません．

yasuohayashibara, 2018-09-08:
ADIS16470の動作の様子です．
http://wiki.ros.org/adi_driver

以下，ADIS16460シリーズに関する議論もされていました．
https://github.com/tork-a/adi_driver/issues/10

Hiroki-Goto, 2018-09-10:
柿沼さんのアドバイスを参考に、感度の調整を行ったあとの動作結果
https://drive.google.com/open?id=1WNCXVoIzDQDtgK4_5P6YYLnBIDge9_Zi

yasuohayashibara, 2018-09-11:
同じ動画をYoutubeにもアップロードしました．
https://youtu.be/3h07mi7UVEQ

あとは実機で試したいと思います．

daiki-shibahara, 2018-09-11:
αに取り付けられるか確認する

Hiroki-Goto, 2018-09-12:
取り付け版を作成しαに取り付けた
#372 にて動作確認を行う
[image]

Hiroki-Goto, 2018-09-20:
WIKIを追加し終わったら閉じる

Hiroki-Goto, 2018-09-21:
wikiを追加した
ただ、[adi_driver](http://wiki.ros.org/adi_driver)の中身を書き換える必要があるのでforkする必要があるかも・・・

Hiroki-Goto, 2018-09-24:
open-rdcにforkしました

closed 2018-10-18 by Hiroki-Goto