open-rdc/orne_navigation #372: 提供されたIMUを使用し自律移動

buzzshima, 2018-09-11:
動作確認が終了したので，実機に取り付けて既存ものとの比較．(amclの回転のパラメータとekfののパラメータの変更したほうがより推定が頑健になる可能性がある)

yasuohayashibara, 2018-09-12:
実験の様子の動画です．
2号館3Fを3周して開始と同じ方向にロボットを向けて終了しました．
https://youtu.be/vmSNoaiCAwo

@Hiroki-Goto IMUのデータをお願いします．

Hiroki-Goto, 2018-09-13:
https://drive.google.com/open?id=1n8wfvDi5QmOg83N_mRy2xyuPhJBsXD0i　
にIMUのデータ（クオタニオンとrpyの２種類）をアップしました

yasuohayashibara, 2018-09-14:
グラフ化しました．
[image]

時間的には4分半ですが，かなり信頼性の高い値が計測できていると思います．

JinyaIseki, 2018-10-04:
共分散行列を一万分の一でもいいかも

yasuohayashibara, 2018-10-04:
美濃君と行っていた研究において，gmappingで用いた値ですが，以下のような小さい値が良い地図が描けました．

```
      <param name="srr" value="0.000001"/>
      <param name="srt" value="0.000002"/>
      <param name="str" value="0.000001"/>
      <param name="stt" value="0.000002"/>
```

https://github.com/open-rdc/create-map-with-landmark/issues/6

Hiroki-Goto, 2018-10-11:
統合している箇所 combine drの共分散行列を０変更

buzzshima, 2018-12-06:
自律走行して問題がなかったため終了

closed 2018-12-06 by buzzshima