open-rdc/orne_navigation #373: 2018年度結界地図を作成

buzzshima, 2018-09-11:
侵入禁止エリアをgimpで使って黒く塗りつぶす

ToshihiroNagao, 2018-09-19:
公園内はまだ作成中です。

ToshihiroNagao, 2018-10-04:
公園内の地図の合成が終了後、結界を張ります。

daiki-shibahara, 2018-10-09:
https://drive.google.com/file/d/1tZYNv8spjAQ1Rbtjy99bBbCRV0-OwBWU/view?usp=sharing
の地図に結界を貼ったものです
https://drive.google.com/file/d/1PXvaR26tDQbMTf4GJH3LlBpwIjcjxWxN/view?usp=sharing

ToshihiroNagao, 2018-10-11:
10月13日、14日の試走会で使用する地図についてですが、Google Driveの共有フォルダに上げます。リンクを載せておきます。
https://drive.google.com/drive/folders/1Mz6iW1A2g99bWVOACE77C1IQ0uoCNtIt?usp=sharing

ToshihiroNagao, 2018-10-25:
10月13日、14日の試走会で使用した公園内の地図ですが、修正を行いました。今後はこの地図を元に結界を貼っていく予定です。
[image]

ToshihiroNagao, 2018-10-25:
下の部分が切れていたので、修正しました。今後はこの地図を元に結界を貼っていく予定です。
[image]

ToshihiroNagao, 2018-11-03:
11月4日の試走会で使用するbetaの地図についてですが、Google Driveの共有フォルダに上げます。リンクを載せておきます。
https://drive.google.com/drive/folders/1WQphvMUGrdYuNjIkb5_EHjPUnNLmGyc9?usp=sharing
試走会終了後、修正するべきものがあれば修正していきたいと思います。

ToshihiroNagao, 2018-11-06:
公園内の地図と結界張りを現在、大至急修正を行っております。
現時点ではこのような状態です。
[image]

ToshihiroNagao, 2018-11-07:
結界張りを完了しました。
現在はGoogleドライブにあります。リンクは以下の通りです。
https://drive.google.com/drive/folders/1WQphvMUGrdYuNjIkb5_EHjPUnNLmGyc9

ToshihiroNagao, 2018-11-08:
向きをそろえて完成しました。issueを閉じます。

closed 2018-11-08 by ToshihiroNagao