open-rdc/orne_navigation #376: 早急に行うタスク

Hiroki-Goto, 2018-09-20:
・チェックシート
・持ち物リスト
・購入品の確認

Hiroki-Goto, 2018-09-21:
チェックシート作成完了
置き場所 https://drive.google.com/open?id=1taetjefbDHvtQvALUUBnaJY8bge_5d6g

Hiroki-Goto, 2018-09-26:
持ち物リストを更新しました

daiki-shibahara, 2018-09-29:
αの安全装置が断線していたため取付し直した
動作確認済みです
[image]

250型の圧着端子が無くなった為，発注お願いします
ミスミの方確認したら丸形しかなく　https://jp.misumi-ec.com/vona2/result/?Keyword=250+%E5%9C%A7%E7%9D%80%E7%AB%AF%E5%AD%90&isReSearch=1

今まで使っていたのは　https://www.customjapan.net/shop/g/g13214234/
こちらです

平型のオスメスか丸形のオスメスどちらが良いでしょうか？

yasuohayashibara, 2018-09-29:
ミスミでもあります．
https://jp.misumi-ec.com/vona2/result/?seriesPage=1&Keyword=%E5%B9%B3%E5%9E%8B%E7%AB%AF%E5%AD%90

あと，過去に平型端子を調達したことがこちらのissueに残っていますが，そのときは夏迫君が買いに行ったと思われます．林原の発注履歴にはありません．

daiki-shibahara, 2018-10-01:
amazonで見つけました
発注お願いします
https://www.amazon.co.jp/%E3%82%A8%E3%83%BC%E3%83%A2%E3%83%B3-%E5%B9%B3%E5%9E%8B%E7%AB%AF%E5%AD%90%E3%82%BB%E3%83%83%E3%83%88-250%E5%9E%8B-30%E3%82%BB%E3%83%83%E3%83%88-1156/dp/B001VNW3JQ?th=1

yasuohayashibara, 2018-10-01:
購入しました．

> amazonで見つけました
> オスメスの6セット発注お願いします

closed 2018-10-04 by JinyaIseki

yasuohayashibara, 2018-10-30:
@daiki-shibahara 
検収して，芝原君の机の上に置きました．

> 購入しました．
> 
> > amazonで見つけました
> > オスメスの6セット発注お願いします