open-rdc/orne_navigation #381: TOP-URG用の電源基盤の選定

Hiroki-Goto, 2018-10-18:


Norihiro02, 2018-10-18:
メモ:
入力24[Ⅴ]
出力12[V]

Norihiro02, 2018-10-18:
とりあえず候補
+ 表面実装ＤＣ－ＤＣコンバーター（５Ｖ／１２Ｖ切替・２Ａ出力）ＭＰＤ６Ｍ００７Ｓ
    + http://akizukidenshi.com/catalog/g/gM-06515/
+ dcdc　降圧モジュール　DROK DC 24V / 12V / 5V　降圧降圧電源　モジュール　8-35V〜1.5-24V　可変コンバータ
    + http://u0u1.net/MFQX

yasuohayashibara, 2018-10-18:
村田製作所で手頃そうと考えましたが，外付け部品が必要なので少々大変です．
選定したらメンションつけてコメントしてください．

> 村田：表面実装ＤＣ－ＤＣコンバーター（５Ｖ／１２Ｖ切替・２Ａ出力）ＭＰＤ６Ｍ００７Ｓ
> * http://akizukidenshi.com/catalog/g/gM-06515/

Norihiro02, 2018-10-19:
以下のモジュールを購入していただきました。
ありがとうございました。
https://www.marutsu.co.jp/pc/i/826652/

yasuohayashibara, 2018-10-23:
到着しました．
林原が持っていますので，できれば取りに来てください．

Norihiro02, 2018-10-26:
電源モジュールのテスト終わりました。
[image]
↑動作確認風景

[image]
↑配線図

Hiroki-Goto, 2018-11-01:
URGのコネクタの取り付け

Norihiro02, 2018-11-04:
実際にURGに接続し、動作確認しました。

ToshihiroNagao, 2018-11-08:
防水をする

closed 2018-11-08 by Norihiro02