open-rdc/orne_navigation #393: ドキュメント更新

buzzshima, 2018-12-06:
中身の変更に伴いなドキュメントの編集

Hiroki-Goto, 2019-02-13:
変更するドキュメント

- [ ] readme

- [ ] wiki（日本語、英語版）
  - suspend関連
  - 反射的な回避動作

追加するドキュメント
- [ ] GNSS関連

- [ ] graph_navigation

- [ ] パラメータ周り？

Hiroki-Goto, 2019-03-21:
wikiの更新（日本語、英語の両方）を行った
readmeに関しては#396 でプルリクを送った

@buzzshima サスペンドのwikiって作成しますっけ？
@Norihiro02  graph_navigationのwikiの作成をお願いします（作成しない場合はコメントお願いします）

MasayaOkada, 2019-03-28:
wikiを見て環境構築を行ったところ、ira_laser_tools(βのurgeをつなぐパッケージ)と orne_sayを消してからcatkin buildをすると環境構築ができた。

yasuohayashibara, 2019-03-30:
`ira_laser_tools`のbranchを`kinekic`にすることが必要のようです．

```
cd ~/catkin_ws/src/ira_laser_tools
git checkout kinetic
```

MasayaOkada, 2019-04-04:
ira_laser_toolsのブランチを変えたらうまくいくことが確認できました。あと、orne_sayは消さなくてもビルド通りました。

NaokiSato102, 2019-10-29:
当方の環境でもira_laser_toolsのブランチを変えたら無事完了しました。
ypspur.hの件が発生しましたがWiki通り行ったところ上記の通り無事完了しました。

taishiyamamoto, 2020-05-31:
melodic上ですが下記の変更を行ったところ，ypspur.hが見つからない問題はなくなりました．
https://github.com/open-rdc/icart/commit/79a2df60683c003c4d06003a3bd6a3023eb6ea4e#diff-74e2b2c150a093c296060969eb0fe7a9

taishiyamamoto, 2020-06-16:
graph_navigationのパッケージ
https://github.com/Norihiro02/graph_navi

taishiyamamoto, 2020-07-06:
graph_navigationのドキュメントを書きました。
https://github.com/open-rdc/graph_navi

taishiyamamoto, 2020-07-06:
パラメータ周りに関するドキュメントを書きました。
https://github.com/open-rdc/orne_navigation/wiki/%E3%83%91%E3%83%A9%E3%83%A1%E3%83%BC%E3%82%BF%E3%83%81%E3%83%A5%E3%83%BC%E3%83%8B%E3%83%B3%E3%82%B0%E3%81%AB%E9%96%A2%E3%81%97%E3%81%A6

taishiyamamoto, 2020-07-06:
反射的な回避動作って以下のissueの内容が関係している？
https://github.com/open-rdc/orne_navigation/issues/339
おそらく以下のブランチで開発していたと思われます。
https://github.com/open-rdc/orne_navigation/tree/test_%23339

taishiyamamoto, 2020-07-07:
反射的な回避動作のリポジトリありました．
https://github.com/open-rdc/avoid_clashing

このリポジトリのREADMEにアルゴリズムと実行方法が書かれています．

taishiyamamoto, 2020-07-07:
反射的な回避動作のリンクをwikiに記載しました．
https://github.com/open-rdc/orne_navigation/wiki/%E5%9B%9E%E5%BE%A9%E8%A1%8C%E5%8B%95%E3%81%AB%E9%96%A2%E3%81%97%E3%81%A6

taishiyamamoto, 2020-07-07:
@yasuohayashibara 
ドキュメントの記載が一通り完了したのでcloseしても大丈夫でしょうか？

taishiyamamoto, 2020-07-07:
ドキュメントの作成が一通り完了したのでcloseします．

NaokiSato102, 2020-08-13:
Kinetic orne環境が飛びまして、再構築しようとしたんですけど、ビルドが通らないです。

> `ira_laser_tools`のbranchを`kinekic`にすることが必要のようです．
> 
> ```
> cd ~/catkin_ws/src/ira_laser_tools
> git checkout kinetic
> ```
これはやったんですけど、icartのほうがなんか、
```
fatal error: ypspur.h: そのようなファイルやディレクトリはありません
```
と成ります。
melodic移行するというのも手ですが研究の作業量が増殖するので辛いです。

yasuohayashibara, 2020-08-13:
インストールのドキュメントに解決法がありますよ．

NaokiSato102, 2020-08-13:
```
find_pacage(YPSpur)をfind_pacage(ypspur)，
またはfind_pacage(ypspur)をfind_pacage(YPSpur)
```
find_packageでは？
タイポか？環境差か？

あと、何度もBuild通してもダメでしたが、思い立って先に

```
catkin build icart_mini_driver 
```
と打ち込んで何回かBuildしたあとにもう一度普通にやったら、うまく行きました。
makeが一部タイミングがかち合ってるのかもしれません。
要検証・・・？

taishiyamamoto, 2020-08-13:
上記のエラーに対応するプルリクを投げました。
https://github.com/open-rdc/icart/pull/126
https://github.com/open-rdc/orne_navigation/pull/490

@yasuohayashibara 
よろしければレビューの後、マージお願いします。
@NaokiSato102 
できたら動作確認してください。

NaokiSato102, 2020-08-13:
> できたら動作確認してください。
もうビルド通っちゃったんでどうすればいいんでしょう。消す？

taishiyamamoto, 2020-08-13:
できたらブランチを切り替えてビルドし直してみてください。

NaokiSato102, 2020-08-13:
ros疎いんでちょっと具体的な捜査がどうするものかわからないのですけど、一度適当にBranchを切り替えてエラーにさせた後に正しいのに戻すとかそういう事ですか？

taishiyamamoto, 2020-08-13:
ビルドにcatkin buildを使っていればcatkin cleanというコマンドで実行ファイルをまとめて消せます。

その後、git checkoutで該当するブランチに切り替えた状態でビルドしてもらえるとありがたいです。

NaokiSato102, 2020-08-13:
了解です。

NaokiSato102, 2020-08-13:
`error: pathspec 'fix_cmake' did not match any file(s) known to git.`
となりました。
[リポジトリに存在するはずのリモートブランチが表示されない - KAIZEN TECH BLOG](https://kaizentech.hatenablog.com/entry/2018/06/26/093756)
[gitでどうしてもリモートのブランチがcheckoutできないとき - Qiita](https://qiita.com/hasht/items/bbdd1763aae7f9f614ee)
[git fetchしてgit branch -rしてもリモートブランチが表示されなかった時 - Qiita](https://qiita.com/chro96/items/69c5be95729fef5622ca)
を参考にすれば解決できそうですがこの現象に遭遇したことはありますか？

taishiyamamoto, 2020-08-16:
git pullでリモートリポジトリと同期しましたか？

NaokiSato102, 2020-08-16:
実は開発環境がまたぶっ飛んだ（カーネルのせいみたい）ので再構築環境ではまだそこにたどり着いてないです。また同じラインに立ち次第バグフィックス流しますね。

NaokiSato102, 2020-08-30:
環境再構築がやっとこさ済んだのですが、何をやればよかったのでしょうか?

とりあえず、[これ](https://github.com/open-rdc/icart/pull/126)を試しました。
> `error: pathspec 'fix_cmake' did not match any file(s) known to git.`
> となりました。fetchしてないとかそういうちゃちな問題ではないです。
> [リポジトリに存在するはずのリモートブランチが表示されない - KAIZEN TECH BLOG](https://kaizentech.hatenablog.com/entry/2018/06/26/093756)
> [gitでどうしてもリモートのブランチがcheckoutできないとき - Qiita](https://qiita.com/hasht/items/bbdd1763aae7f9f614ee)
> [git fetchしてgit branch -rしてもリモートブランチが表示されなかった時 - Qiita](https://qiita.com/chro96/items/69c5be95729fef5622ca)
> を参考にすれば解決できそうですがこの現象に遭遇したことはありますか？

ずっとorne_naviのブランチかと思ってましたが、icert_miniの方だと気づきました。
checkout後、buildをしましたが、エラーが起きました。

> CMake Error at /home/naoki/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt:18 (find_package):
>   By not providing "Findypspur.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "ypspur", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "ypspur" (requested
>   version 1.17.0) with any of the following names:
> 
>     ypspurConfig.cmake
>     ypspur-config.cmake
> 
>   Add the installation prefix of "ypspur" to CMAKE_PREFIX_PATH or set
>   "ypspur_DIR" to a directory containing one of the above files.  If "ypspur"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> cd /home/naoki/catkin_ws/build/icart_mini_driver; catkin build --get-env icart_mini_driver | catkin env -si  /usr/bin/cmake /home/naoki/catkin_ws/src/icart_mini/icart_mini_driver --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/naoki/catkin_ws/devel/.private/icart_mini_driver -DCMAKE_INSTALL_PREFIX=/home/naoki/catkin_ws/install; cd -

NaokiSato102, 2020-09-01:
icart_miniのを先にビルド通してからやると何度もビルド通し直しても通らなかったのが素直に通るようになったので、並行ビルドの順序問題かもしれません。

NaokiSato102, 2020-09-04:
>     catkin build icart_mini_driver 
> と打ち込んで何回かBuildしたあとにもう一度普通にやったら、うまく行きました。
> makeが一部タイミングがかち合ってるのかもしれません。
> 要検証・・・？

と以前書きましたが、今環境では全然うまくいかないです。
open-rdc/icart#126 , #490 を再度検証しましたが、やはり検証時に前とおんなじ、「Could not find a package configuration file provided by "ypspur" (requested
version 1.17.0) with any of the following names:」が出ました。
従来方法も試したところこれはうまくいきました。しかしエラー文が興味深いです。
```
Warnings   << icart_mini_driver:check /home/naoki/catkin_ws/logs/icart_mini_driver/build.check.007.log
CMake Warning at /home/naoki/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt:24 (find_package):
  By not providing "Findypspur.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ypspur", but
  CMake did not find one.

  Could not find a package configuration file provided by "ypspur" with any
  of the following names:

    ypspurConfig.cmake
    ypspur-config.cmake

  Add the installation prefix of "ypspur" to CMAKE_PREFIX_PATH or set
  "ypspur_DIR" to a directory containing one of the above files.  If "ypspur"
  provides a separate development package or SDK, be sure it has been
  installed.

cd /home/naoki/catkin_ws/build/icart_mini_driver; catkin build --get-env icart_mini_driver | catkin env -si  /usr/bin/make cmake_check_build_system; cd -
```
エラーが、open-rdc/icart#126 , #490 の時とほとんど一緒なんですよ。差が解らん。

tiger0421, 2020-09-04:
https://github.com/open-rdc/orne_navigation/wiki/Install_ORNE_pkg#4install-ypspur
ypspurをcloneした後に一度ビルドしてから，5番のORNEのインストールをするとどうなりますかね？

NaokiSato102, 2020-09-04:
つまりはどういう事をするんでしょう？
catkin cleanして、
catkin_wsを全部削除して、ypspurをcloneした後に一度ビルドしてから，5番のORNEのインストールをする？

catkin cleanして、すでにcloneされているypspurを一度ビルドしてから、
$ wstool init
$ wstool merge orne_navigation/orne_pkgs.install
$ wstool up
して、
$ wstool init
$ wstool merge orne_navigation/orne_pkgs.install
$ wstool up
して、
rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
したあとに、
catkin build
しても、有意差はありませんでしたよ。意図した処理で無かった場合もう少し細かく説明お願いします。

NaokiSato102, 2020-09-04:
現状の知識を活かしてセットアップscriptの改良案を出すとなると以下の通りになります。

```
#!/bin/bash

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
catkin_init_workspace
cd ~/catkin_ws/
catkin build
cd ~/catkin_ws/src
git clone https://github.com/DaikiMaekawa/ypspur.git
git clone https://github.com/open-rdc/orne_navigation
wstool init
wstool merge orne_navigation/orne_pkgs.install
wstool up
rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

catkin build icart_mini_driver
sed -i -e 's/find_package(ypspur)/find_package(YPSpur)/g' ~/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt
catkin build icart_mini_driver
sed -i -e 's/find_package(YPSpur)/find_package(ypspur)/g' ~/catkin_ws/src/icart_mini/icart_mini_driver/CMakeLists.txt
catkin build icart_mini_driver
catkin build icart_mini_driver
catkin build

echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
exit
```
改良点はypspurのbuild試行時にほかのパッケージごとでなく、問題点たるicart_mini_driverのみbuild試行することで所要時間の短縮を図っています。

tiger0421, 2020-09-04:
> catkin_wsを全部削除して、ypspurをcloneした後に一度ビルドしてから，5番のORNEのインストールをする？
それであってる．

tiger0421, 2020-09-04:
そしたら，
ypspur/cmake/YPSpurConfig.cmake
を
ypspur/cmake/ypspurConfig.cmake
にリネームしてからビルドするとどうなる？

NaokiSato102, 2020-09-04:
> > catkin_wsを全部削除して、ypspurをcloneした後に一度ビルドしてから，5番のORNEのインストールをする？
> 
> それであってる．

それだけではだめだった。

NaokiSato102, 2020-09-04:
ダメだった後に
```
そしたら，
ypspur/cmake/YPSpurConfig.cmake
を
ypspur/cmake/ypspurConfig.cmake
にリネームしてからビルドするとどうなる？

ypspur/CMakeLists.txt内の
install(FILES cmake/YPSpurConfig.cmake DESTINATION share/${PROJECT_NAME}/cmake/)
を
install(FILES cmake/ypspurConfig.cmake DESTINATION share/${PROJECT_NAME}/cmake/)
に変更

ypspur/cmake/YPSpurConfig.cmake内の
find_package_handle_standard_args(YPSpur DEFAULT_MSG
を
find_package_handle_standard_args(ypspur DEFAULT_MSG
に変更
```
をやったが、ダメだった。

NaokiSato102, 2020-09-05:
[さっきのダメ](https://github.com/open-rdc/orne_navigation/issues/393#issuecomment-687296644)の詳細
rm ~/catkin_ws -rf
mkdir -p catkin_ws/src
catkin build

cd ~/catkin_ws/src
git clone https://github.com/DaikiMaekawa/ypspur.git
catkin build ypspur

 git clone https://github.com/open-rdc/orne_navigation
wstool init
wstool merge orne_navigation/orne_pkgs.install
wstool up

rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

catkin build

とやったが駄目だった。

NaokiSato102, 2020-09-05:
> ```
> ypspur/cmake/YPSpurConfig.cmake
> を
> ypspur/cmake/ypspurConfig.cmake
> ```
> 
> にリネームしてからビルドして、ダメだった後に
> 
> ```
> ypspur/CMakeLists.txt内の
> install(FILES cmake/YPSpurConfig.cmake DESTINATION share/${PROJECT_NAME}/cmake/)
> を
> install(FILES cmake/ypspurConfig.cmake DESTINATION share/${PROJECT_NAME}/cmake/)
> に変更
> 
> ypspur/cmake/YPSpurConfig.cmake内の
> find_package_handle_standard_args(YPSpur DEFAULT_MSG
> を
> find_package_handle_standard_args(ypspur DEFAULT_MSG
> に変更
> ```
> 
> を実行するって理解したんですけど、正しい理解ですか？それとも全部同時にやる？

そもそも全部同時にやらないとエラーが起きる。

NaokiSato102, 2020-09-05:
```
catkin clean -y
cd
rm ~/catkin_ws -rf
mkdir -p catkin_ws/src
cd ~/catkin_ws
catkin build

cd ~/catkin_ws/src
git clone https://github.com/DaikiMaekawa/ypspur.git

cd ~/catkin_ws/src/ypspur/cmake/
mv YPSpurConfig.cmake ypspurConfig.cmake
sed -i -e 's/YPSpur/ypspur/g' ~/catkin_ws/src/ypspur/cmake/ypspurConfig.cmake
sed -i -e 's/YPSpurConfig/ypspurConfig/g' ~/catkin_ws/src/ypspur/CMakeLists.txt

catkin build ypspur

cd ~/catkin_ws/src
git clone https://github.com/open-rdc/orne_navigation
wstool init
wstool merge orne_navigation/orne_pkgs.install
wstool up

rosdep install --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

catkin build

```
これやってもダメだった。

NaokiSato102, 2020-09-05:
エラーメッセージに従い、
`env CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/home/naoki/catkin_ws/src/ypspur catkin build`
で無理やりypspurにパスを通してbuildしてみたが駄目だった。
もっと良い方法、ちゃんとした方法があるかもしれない。

NaokiSato102, 2020-09-08:
#495 へ継承しclose

closed 2020-09-08 by NaokiSato102