open-rdc/orne_navigation #394: velodyneを用いたMCLのリセットパラメータの調整

buzzshima, 2018-12-06:
頻繁にリセットが起きるので調整する

ToshihiroNagao, 2019-02-14:
引継ぎを行いましたので、担当者を変更します。

atsushi10mino, 2019-04-23:
撮影した動画とパラメータをのせる

hoshianaaa, 2019-05-28:
0.001 [https://www.youtube.com/watch?v=darooOaQ5Fw](url)
0.002 [https://www.youtube.com/watch?v=hMn1pB1Q1XQ](url)
0.003 [https://www.youtube.com/watch?v=MfNCDsPO-FY](url)
0.004 [https://www.youtube.com/watch?v=i77ORQtg8Bc](url)
0.005 [https://www.youtube.com/watch?v=gSK6XqKA4yM](url)
0.006 [https://www.youtube.com/watch?v=dwg9xNh6DVk](url)
0.007 [https://www.youtube.com/watch?v=OLoNjVU0EKg](url)
0.008 [https://www.youtube.com/watch?v=LCWAtFZfc_g](url)

hoshianaaa, 2019-05-28:
0.001 https://www.youtube.com/watch?v=darooOaQ5Fw&t=71s

hoshianaaa, 2019-05-28:
0.001 https://www.youtube.com/watch?v=darooOaQ5Fw&t=71s
0.002 https://www.youtube.com/watch?v=hMn1pB1Q1XQ
0.003 https://www.youtube.com/watch?v=MfNCDsPO-FY&t=13s
0.004 https://www.youtube.com/watch?v=i77ORQtg8Bc

hoshianaaa, 2019-05-28:
0.005 https://www.youtube.com/watch?v=gSK6XqKA4yM
0.006 https://www.youtube.com/watch?v=dwg9xNh6DVk
0.007 https://www.youtube.com/watch?v=OLoNjVU0EKg
0.008 https://www.youtube.com/watch?v=LCWAtFZfc_g

JinyaIseki, 2019-05-28:
誘拐問題を発生させ、実験を行う

hoshianaaa, 2019-06-03:
[https://www.youtube.com/watch?v=C5AsjfL-WFc](url)

hoshianaaa, 2019-06-03:
https://www.youtube.com/watch?v=C5AsjfL-WFc

hoshianaaa, 2019-06-06:
アドバイスを受け、つくばの環境に近い図書館前で調整を行った。
0.002,0.003はパーティクルが発散
0.001でパーティクルが発散しなくなった。

hoshianaaa, 2019-06-06:
動画
0.001:https://www.youtube.com/watch?v=vlxj5OSpE0w(0:47 ~ 1:17 )
0.002:https://www.youtube.com/watch?v=89fqQfcVjRU(0:26 ~ 0:50 )
0.003:https://www.youtube.com/watch?v=Kx7OCuJGWJc(0:40 ~ 1:15)

hoshianaaa, 2019-06-06:
他にリセットの起こりそうな場所で試した
場所は駐輪場前の芝生
この場所では、0.001でも大きいかもしれない

0.001：https://www.youtube.com/watch?v=XymxQThBK5o (0:36 ~ 1:08 )
0.0009：https://www.youtube.com/watch?v=8rhYYHUEnR0 (0:24 ~ 0:50)

daiki-shibahara, 2019-07-02:
茜浜公園や新習志野校舎で試してみる

Akito-Shirai, 2019-08-27:
とりあえず0.001で実行し、それから変えていく。

closed 2019-11-26 by taishiyamamoto